hku-mars/livox_camera_calib #3: Process dies on LiDAR edge extraction

VisionaryMind, 2021-03-31:
I have setup an environment with Ubuntu 18.04 and ROS Melodic. All dependencies have been installed, however, using your sample ROS bag files, I am not able to get past the Lidar edge extraction phase. Here is the terminal output:

```
Camera Matrix: 
[1364.45, 0, 958.327;
 0, 1366.46, 535.074;
 0, 0, 1]
Distortion Coeffs: 
[0.0958277;
 -0.198233;
 -0.000147133;
 -0.000430056;
 0]
Init extrinsic: 
[0, -1, 0, 0;
 0, 0, -1, 0;
 1, 0, 0, 0;
 0, 0, 0, 1]
[ INFO] [1617231323.709287053]: Loading the rosbag /home/visionarymind/Documents/scans/livoxcalib/outdoor1.bag
[ INFO] [1617231328.493049027]: Sucessfully load Point Cloud and Image
[ INFO] [1617231328.493564736]: Building Voxel
[ INFO] [1617231357.841487217]: Extracting Lidar Edge
[lidar_camera_calib-2] process has died [pid 2586, exit code -11, cmd /home/visionarymind/vision_ws/devel/lib/livox_camera_calib/lidar_camera_calib /home/visionarymind/vision_ws/src/livox_camera_calib/config/camera.yaml /home/visionarymind/vision_ws/src/livox_camera_calib/config/config_outdoor.yaml /home/visionarymind/Documents/scans/livoxcalib/outdoor1.bag /home/visionarymind/vision_ws/src/livox_camera_calib/result __name:=lidar_camera_calib __log:=/home/visionarymind/.ros/log/2c3b8fe2-9274-11eb-9545-000c29df74c2/lidar_camera_calib-2.log].
log file: /home/visionarymind/.ros/log/2c3b8fe2-9274-11eb-9545-000c29df74c2/lidar_camera_calib-2*.log
```
Would you have any advice on how to correct this problem?

ChongjianYUAN, 2021-04-01:
> I have setup an environment with Ubuntu 18.04 and ROS Melodic. All dependencies have been installed, however, using your sample ROS bag files, I am not able to get past the Lidar edge extraction [...]

What is the PCL version you are using? Maybe you should use PCL1.7

VisionaryMind, 2021-04-01:
> What is the PCL version you are using? Maybe you should use PCL1.7

We were able to get this working by installing a virtual machine with Ubuntu 16.04 and PCL 1.7, so it is clear that later PCL versions can be a problem.

You mentioned in the specs that this can work, however, on Ubuntu 18.04, but Bionic's repo only has PCL 1.8, and it is not possible to compile and install PCL 1.7 on Bionic due to other dependency problems with Eigen, Boost and FLANN, for example. If you have an 18.04 system working, please be so kind as to share your config (conda list or pip freeze). It could be we just need to align the dependencies properly.

For now, we are fine using Ubuntu 16.04 / PCL 1.7, but I would be interested in better understanding how to port this amazing tool over to newer platforms with more up-to-date dependencies. I wish to congratulate you on this superb work. It is one of the most innovative approaches to extrinsics calibration to date and will help our work / research immensely. Thank you for sharing your research with the community.

ChongjianYUAN, 2021-04-01:
> We were able to get this working by installing a virtual machine with Ubuntu 16.04 and PCL 1.7, so it is clear that later PCL versions can be a problem.
>
> You mentioned in the specs that this can [...]

Thank you for your attention. The program will be compatible with the latest version of PCL in the near future, expected in one or two weeks.

closed 2021-04-05 by VisionaryMind

sheng00125, 2022-05-10:
I meet the same problem on ubuntu18 + pcl1.8.

chengfzy, 2022-08-04:
same problem!

Chatoyant19, 2023-06-02:
I solved this prolem. just modified pcl version to the sysytem default version. eg, ubuntu 20.04 + pcl-1.10

zayedrmdn, 2024-08-16:
@Chatoyant19 what did you modified exactly?

martin-liao, 2024-11-19:
I have tried Ubuntu20.04+PCL 1.10 + Ceres 2.0.0. Everything is fine.