hku-mars/livox_camera_calib #4: Inaccurate results, even when providing accurate, manually calibrated extrinsics

VisionaryMind, 2021-04-03:
We were very impressed with the results seen running your sample datasets, but once we tried using our own data, the extrinsics calibration failed in 99% of tested scenarios. In every case, your tool miscalculates scene edges and shifts the image projection, in some cases, by as much as 50-60% off original estimates.

For example, here is the projection of an ultra-HiRes image onto a cloud generated with the Livox Avia:
[image]

And here is the same cloud / image pair filtered through your calibration tool:
[image]

We provided the same exact intrinsics and initial extrinsics that were used to create the first project, hoping that your tool would properly recognize scene edges and perfect / optimize the calibration. Instead, it identifies edges far away from their origin and attempts to shift the projection to match them. This results in a completely inaccurate calibration.

If there are any settings you could recommend that might improve our result, please let me know. I presume that the reason your sample datasets work better is because they are based on scenes with very clean, broad lines, such as those found in artificial structures. We even tried indoor scenes with Aruco / checkerboard plates places in various locations, and your tool does not recognize any of the lines in these plates. It completely ignores the plates and, as with the example above, "hunts" for edges far from the original extrinsic origin.

Perhaps this tool is only applicable in industrial / city-centric settings where large buildings or artificial structures are present. We will attempt to re-test in a "cleaner" environment, but the fact that this tool cannot work with even standard Aruco / checkerboard patterns to properly orient itself makes it mostly unusable for the majority of contexts we would require.

ChongjianYUAN, 2021-04-04:
> We were very impressed with the results seen running your sample datasets, but once we tried using our own data, the extrinsics calibration failed in 99% of tested scenarios. In every case, your tool miscalculates scene edges and shifts the image projection, in some cases, by as much as 50-60% off original estimates.
> 
> For example, here is the projection of an ultra-HiRes image onto a cloud generated with the Livox Avia:
> 
> [image]
> 
> And here is the same cloud / image pair filtered through your calibration tool:
> 
> [image]
> 
> We provided the same exact intrinsics and initial extrinsics that were used to create the first projection, hoping that your tool would properly recognize scene edges and perfect / optimize the calibration. Instead, it identifies edges far away from their origin and attempts to shift the projection to match them. This results in a completely inaccurate calibration.
> 
> Here is the planner cloud:
> 
> [image]
> 
> And the residuals seem to show that the lines are properly detected:
> 
> [image]
> 
> Nonetheless, the optimization shifts everything, producing the result I have posted above. If there are any settings you could recommend that might improve our result, please let me know. I presume that the reason your sample datasets work better is that they are based on scenes with very clean, broad lines, such as those found in artificial structures. We even tried indoor scenes with Aruco / checkerboard plates placed in various locations, and your tool does not recognize any of the lines in these plates. It completely ignores the plates and, as with the example above, "hunts" for edges far from the original extrinsic origin.
> 
> Perhaps this tool is only applicable in industrial / city-centric settings where large buildings or artificial structures are present. We will attempt to re-test in a "cleaner" environment, but the fact that this tool cannot work with even standard Aruco / checkerboard patterns to properly orient itself makes it mostly unusable for the majority of contexts we would require.

You can read our related paper before you test our code on your own sensor set. We analyzed that for more accurate calibration, we need depth-continuous edges. It is obvious that there are not enough depth-continuous in your dataset. Besides, please do not select the scene with excessive texture such as trees and grass. Our algorithm allowed some grass and trees in the scene, but there were too many grass and trees in the data set you provided, so the edge extracted by the camera would not be accurate. You can see that the blue curve in the residual image was very numerous and disorderly. Other people tested our code on the Horizon LiDAR and it worked very well, so a good scene is a prerequisite for a successful calibration.

VisionaryMind, 2021-04-05:
Thank you kindly for the explanation. We did read your paper, but there was some confusion regarding the types of scenes that produce bad results. We attempted indoor scenes with sufficient depth-continuous regions, but there were also depth-discontinuous (for example, a staircase with metal bars along the hand-rail). Now, re-reading your paper, it is clear these scenarios must be avoided:

1. Presence of cylindrical objects.
2. Edge "clusters" with uneven distribution
3. Edges in a single direction (our example above). 
4. Complex textures (trees, grass, stone masonry).

That last subject is a problem for us, because this is a very "green" region with trees and shrubbery everywhere. Many of the buildings in this region, further, are made of stone, creating complex textures that could confuse the algorithm.

So it appears we will need to find a structure that has multiple depth-continuous regions and, optimally, constructed from simple materials such as concrete with simple to no patterns. Again, thank you for taking the time to explain the error in our data collection. I will report back once we have tested other scenes.

VisionaryMind, 2021-04-05:
Our last test was not successful. We located a large building with multiple depth-continuous planes with no trees / foliage. This scene should have been sufficient for your algorithm. Here is the original photo:

[image]

And here is the calibration result:

[image]

Again, the initial alignment was close to perfect --- all major depth-continuous planes and their windows / lines were no more than 2-3cm from the image projection. Nonetheless, the calibration algorithm completely missed all lines. And a looking at the residual this time, it's perfectly clean with close to zero noise.

Would you be able to give us any advice on why this is not working? We are trying now for several days with multiple, varied scenes, and always this same result. What are we doing wrong? Perhaps we should use another tool for camera intrinsic calibration -- the MATLAB tools have consistently generated unpredictable, often inaccurate results.

ChongjianYUAN, 2021-04-06:
> Our last test was not successful. We located a large building with multiple depth-continuous planes with no trees / foliage. This scene should have been sufficient for your algorithm. Here is the [...]

Could you show me the result of LiDAR edge extraction and the projection image?

VisionaryMind, 2021-04-06:
> Could you show me the result of LiDAR edge extraction and the projection image?

We were able to achieve a slightly better result today. The theory that MATLAB does not produce reliable camera intrinsics calibration is correct. MATLAB is extremely inaccurate, and apparently completely mis-calculates distortion. By completely removing the distortion values (resetting them to 0.0), this is the result:

[image]

It is clearly still not accurate, but at least better than before. So it is very possible that what we need is a more accurate set of intrinsics for our DSLR camera. That being said, I also wonder how important the initial extrinsics are for a successful result. Would you be able to provide advice for hand measurements? Currently, the Avia is more or less aligned vertically with the camera sensor and about 16cm above it. The camera has a Leica lens that extends out from the sensor. Here is a picture:

[image]

We attempted to approximate this configuration with the following initial extrinsics:

```
         0.0,  0.0,  -1.0,    0.2,
         1.0,   0.0,    0.0,    -1.0,
         0.0,   0.0,    0.0,    1.0
```

Here is the LiDAR edge extraction:

[image]

And the projection image:

[image]

Hopefully, this scene is adequate, and our task will be to achieve more accurate intrinsic measurement. If you have any other suggestions for how we can improve our result, we would be very appreciative!

ChongjianYUAN, 2021-04-06:
It looks like a pretty good scene and I have two pieces of advice.
1. Why there are still so much noise edge extracted by the camera's image (The blue line in residual image)? Did you change any params about the Canny in the config file? You can see that in the residual image of our dataset, the blue line is more accurate. Maybe you can adjust Canny.gray_threshold(10~30）and Canny.len_threshold(200~500) in config_outdoor.yaml to filter out these noise edges.
2.  If the camera and the LiDAR are positioned in the same direction. The recommended initial extrinsic should be:
[0.0   -1.0   0.0     tx
0.0     0.0   -1.0    ty
1.0     0.0    0.0    tz
0.0     0.0    0.0     0]
where tx,ty tz can be the rough value measured by hand. Actually, in our sensor set, we set tx,ty,tz all to be zero and it still work.

VisionaryMind, 2021-04-06:
> It looks like a good scene and I have two pieces of advice.
> 
> 1. Why there are still so much noise edge extracted by the camera's image (The blue line in residual image)? Did you change any params about the Canny in the config file? You can see that in the residual image of our dataset, the blue line is more accurate. Maybe you can adjust Canny.gray_threshold(10 to 30）and Canny.len_threshold(200 to 500) in config_outdoor.yaml to filter out these noise edges.

Thank you for these recommendations. The noise comes from areas of the building surface that is made of stone. I changed the Canny thresholds as you suggested (to 30 and 500, respectively), and this is the result (edges and projection):

[image]

[image]

[image]

> 2. If the camera and the LiDAR are positioned in the same direction. The recommended initial extrinsic should be:
>    [0.0 ,  -1.0,   0.0 ,    tx
>    0.0,     0.0,   -1.0,    ty
>    1.0 ,    0.0, [...]

We attempted this intrinsic on the result above, only ty must be set to 1, given that our camera is below the LiDAR, not above (as in your scenario). Nonetheless, something seems to be causing this to fail outside the tool's configuration. I can't help but wonder if the intrinsics could be throwing something off. You should know that we aren't able to even use ACSC to calibrate this camera -- it cannot find either 2D or 3D corners. In fact, none of the calibration tools that have worked for us before will work with this camera-LiDAR pair. I do not understand why -- only faulty intrinsics seems possible.

Perhaps it is our version of Ceres-Solver? Our current config:

libceres-dev 1.12.0+dfsg0-5
libceres1 1.12.0+dfsg0-5  

Which version are you using?

ChongjianYUAN, 2021-04-07:
> > It looks like a good scene and I have two pieces of advice.
> > 
> > 1. Why there are still so much noise edge extracted by the camera's image (The blue line in residual image)? Did you change any params about the Canny in the config file? You can see that in the residual image of our dataset, the blue line is more accurate. Maybe you can adjust Canny.gray_threshold(10 to 30）and Canny.len_threshold(200 to 500) in config_outdoor.yaml to filter out these noise edges.
> 
> Thank you for these recommendations. The noise comes from areas of the building surface that are made of stone. I changed the Canny thresholds as you suggested (to 30 and 500, respectively), and this is the result (edges and projection):
> 
> [image]
> 
> [image]
> 
> [image]
> 
> > 1. If the camera and the LiDAR are positioned in the same direction. The recommended initial extrinsic should be:
> >    [0.0 ,  -1.0,   0.0 ,    tx
> >    0.0,     0.0,   -1.0,    ty
> >    1.0 ,    0.0,    0.0,    tz
> >    0.0 ,    0.0,    0.0 ,    0]
> >    where tx,ty tz can be the rough value measured by hand. Actually, in our sensor set, we set tx,ty,tz all to be zero and it still work.
> 
> We attempted this intrinsic on the result above, only ty must be set to 1, given that our camera is below the LiDAR, not above (as in your scenario). Nonetheless, something seems to be causing this to fail outside the tool's configuration. I can't help but wonder if the intrinsics could be throwing something off. You should know that we aren't able to even use ACSC to calibrate this camera -- it cannot find either 2D or 3D corners. In fact, none of the calibration tools that have worked for us before will work with this camera-LiDAR pair. I do not understand why -- only faulty intrinsics seems possible.
> 
> Perhaps it is our version of Ceres-Solver? Our current config:
> 
> libceres-dev 1.12.0+dfsg0-5
> libceres1 1.12.0+dfsg0-5
> 
> Which version are you using?

Probably not Ceres Solver. The version I used is 1.14.  I guess it was a poor initial value that caused the mismatch.  You can measure the relative position between LiDAR and camera by a ruler and set to tx,ty,tz (The unit is m). You say that ty must be set to 1, but actually I didn't think the distance between the camera and the LiDAR in the vertical direction is 1m. Besides, pull the newly version may help you. Finally, accumulate point clouds for 20s or more.

ChongjianYUAN, 2021-04-07:
> Probably not Ceres Solver. The version I used is 1.14. I guess it was a poor initial value that caused the mismatch. You can measure the relative position between LiDAR and camera by a ruler and set [...]

If you still can't solve the problem, you can share the dataset with me (including camera parameters) and I will try to help you.

VisionaryMind, 2021-04-07:
> Probably not Ceres Solver. The version I used is 1.14. I guess it was a poor initial value that caused the mismatch. You can measure the relative position between LiDAR and camera by a ruler and set [...]

I compiled and installed Ceres Solver 1.14 today as well as the latest Eigen, and it had no effect, as you surmised. As for rule measurements, we initially measured Avia above the camera at 17cm and the sensor difference (on Z axis) at ~2cm. Using -0.17 and 0.02, respectively in the initial extrinsics produces a projection that is too low and far away, with worse final results. That is why we were experimenting with must higher values between .5 and 1, even though it does not make logical sense.

I would be pleased to provide you with our dataset, if you will have a moment to troubleshoot. Please indicate which files you need (apart from the Bag and image file) as well as where to upload them. 

We will also attempt to find another scene tomorrow with less shadows, as it is possible the dark areas in the image are preventing optimal edge detection.

ChongjianYUAN, 2021-04-07:
> I compiled and installed Ceres Solver 1.14 today as well as the latest Eigen, and it had no effect, as you surmised. As for rule measurements, we initially measured Avia above the camera at 17cm and [...]

The bag files that contains the point cloud information, the image files and the camera config files. You can directly sent to my email ycj1@connect.hku.hk. Thanks.

VisionaryMind, 2021-04-09:
We attempted another scene today, hoping that the setting would be more compatible with your algorithm. Here is the original photo:

[image]

And here is how your algorithm has interpreted its lines:

[image]

The lines in this capture are very obvious and clean...

[image]

[image]

Without digging into your code, I cannot imagine why it should not be a simple matter to match them, at least within reasonable proximity. Again, the initial projection starts close to the target (with near precision), and your algorithm progressively shrinks and skews it, far from origin. This has been the same behavior for every single scene we have tested.

We have sent you the original dataset. I am skeptical that this algorithm may be applied to high-resolution DSLR cameras. Time-permitting, we will use a low-end option such as the Realsense D455, at much lower resolution. But I do not see how this would make a difference, as our DSLR has close to no distortion, and the intrinsics are proper.

Alexdev585, 2021-04-10:
> We were very impressed with the results seen running your sample datasets, but once we tried using our own data, the extrinsics calibration failed in 99% of tested scenarios. In every case, your [...]
Hi visionarmind

What method did you use to produce the above coloured point cloud?

closed 2021-05-21 by ChongjianYUAN