hku-mars/livox_camera_calib #6: pnp_calib

yanlong658, 2021-05-06:
English version

Thank you for the open source. This work is very perfect.
However, I have the some question about the code.
First, what's the vpnp. I don't know the Physical mean. Could you  provide me reference?
Second, why's the camera frame map into the pixel frame(image frame),  you use the (u0-cx)/ fx.  According to my knowledge, I would use (u0)/fx - cx.
Thank you very much.
[image]

中文版
感謝您提供的開源程式碼,這個程式碼非常的棒
但是我有一些疑問是有關於code的部份
1. 請問什麼是vpnp,我不太清楚其物理意義,請問能麻煩您提供我關於這個參考資料嘛？
2. 在lidar_camera_calib.cpp裡面中,您透過相機內參投影到pixel frame,您使用的是(u0-cx)/ fx,您是先減掉像素中心在除以焦距長. 但照我知識中,應該是(u0)/fx - cx,先除以焦距長,在減掉像素的中心,請問我哪裡搞錯了嘛？

ChongjianYUAN, 2021-05-06:
1. vpnp means add a direction vector to normal pnp problem. You can refer to the following picture
[image]

Then the residual will be:
[image]
here, Π() is the pin-hole projection model, f() is the distortion model.
2.  The process (u0-cx)/fx is to restore x, which will be corrected by distortion model:
[image]

ChongjianYUAN, 2021-05-10:
> English version
> Excuse me, this is the last one problem.
> Why the Vpnp direction is not used the (pi - qi) * normal direction of camera, however you use the V = n_nt, V= I-V, R = V_ R* V.transpose().
> 
> Thanks for your reply
> 
> 中文版
> 你好,感謝您的回覆
> 我還有最後一個問題要麻煩您
> 那就是我理解的vpnp 的residual應該是直接(pi - qi) * 線的normal vector, 但您卻是使用 V = n_nt, V= I-V, R = V_ R* V.transpose()
> 我不太懂其這樣會形成paper上面說的residual
> 
> 感謝您

Sorry for my mistake. The direction n in the code is not the same as the n in my reply. The n in the code is the direction of camera's edge but not the vertical direction vector. If n is the direction of  camera's edge, the residual will be (I-n*nt)r(P-q).  I will correct it in the next version and thank you again for your question.

closed 2021-05-11 by yanlong658