hku-mars/livox_camera_calib #7: velodyne support

gonultasbu, 2021-05-18:
I have been searching for a targetless automatic lidar-camera calibration package and I came across your repository, however the info section on README says that Velodyne support is to be added later.  I was wondering why is it not possible to calibrate Velodyne and RealSense using this package. Theoretically, I can just modify the corresponding topic names before calibration and it should be able to treat the pointcloud as a `sensor_msgs/PointCloud2` object independently (from the hardware). Why is that not possible? I am willing to try it myself in a few days but wanted to consult you before that.

Thank you for your work!

ChongjianYUAN, 2021-05-21:
Hello, I haven't tested the code on Velodyne, but I've tested it on Ouster. As the spinning LiDAR is repeated scanning and can not get a dense point cloud without motion, which will result in an insufficient and inaccurate number of edges extracted.  Hence, I recommend adding the preprocess process(e.g. [Fast-LIO)](https://github.com/hku-mars/FAST_LIO) to make the point cloud of the spinning LiDAR denser. Here's an example:
[image]
After that, our algorithm can work well and get an accurate extrinsic value:
[image]
The source code will be updated shortly.

gonultasbu, 2021-05-24:
That makes sense, thanks.

I was wondering whether it's possible to calibrate using image sequences such as videos to mitigate this problem instead of making a single point cloud image denser with preprocessing.

gonultasbu, 2021-05-24:
Running a mapping algorithm to get a denser LiDAR map indeed makes sense, thanks!

zzkslam, 2021-07-09:
> Running a mapping algorithm to get a denser LiDAR map indeed makes sense, thanks!

Can you tell me which mapping algorithm works?

gonultasbu, 2021-07-10:
> > Running a mapping algorithm to get a denser LiDAR map indeed makes sense, thanks!
> 
> Can you tell me which mapping algorithm works?

I didn't have the time to test them right now but Fast-LIO referred by the author might come useful.

zzkslam, 2021-07-11:
> > > Running a mapping algorithm to get a denser LiDAR map indeed makes sense, thanks!
> > 
> > 
> > Can you tell me which mapping algorithm works?
> 
> I didn't have the time to test them right now but Fast-LIO referred by the author might come useful.

OK，thanks

CanCanZeng, 2021-08-12:
Hi, I'm trying to calibrate a velodyne HDL32 lidar with a camera, as you said, the fast-lio can be used to get denser point cloud, but I find that the point cloud produced by fast-lio is defined in IMU coordinate system. So I need to transform the point from IMU to lidar first and then calibrate the transformed cloud with camera.
Actually, the extrinsic between IMU and lidar is not that accurate, so how can I get an accurate extrinsic between lidar and camera? @ChongjianYUAN

islamtalha01, 2021-10-11:
Hi 
      I have checked your git repository regarding lidar camera calibration. I am a research intern at my Home university NUST Pakistan. I have an os1 64 spinning lidar available at my lab. I need to calibrate the camera and lidar. Now kindly tell me the precise step I need to follow to get that calibration parameter. I don't know how to use Fast LIO for spinning lidar calibration. kindly outline the steps to follow to get extrinsic parameters with your package using os1 64 lidar. I would very thankful to you for this favour.I have bag file recorded with topic pointcloud2 and hikvision camera image. kindly help me in this regard

TianArthur, 2022-05-07:
> Hello, I haven't tested the code on Velodyne, but I've tested it on Ouster. As the spinning LiDAR is repeated scanning and can not get a dense point cloud without motion, which will result in an [...]

Thank you very much for your excellent work.Is there any update on work of adding fastlio for calibrating repetitive ldiar.

ChongjianYUAN, 2022-05-11:
> Thank you very much for your excellent work.Is there any update on work of adding fastlio for calibrating repetitive ldiar.

Sorry, I haven't had enough time recently to open source a version that supports Spinning LiDAR . You can try this repo: https://github.com/AFEICHINA/extended_lidar_camera_calib

kahowang, 2022-05-13:
> 
Thank you very much for your excellent work and reply~

simi-asher, 2023-01-31:
Hi, if anyone is looking for an extension to spinning LiDARs but in ROS2, please check out my repository [here](https://github.com/simi-asher/lidar_camera_calib). Happy to reply to questions or issues.