hku-mars/livox_camera_calib #8: Calibration Problem

yanlong658, 2021-05-19:
English Version
Hello
I have some question about the calibration problem.
I set my vlp-16 and D345i camera as your configuration and I calibrate the extrinsic parameter. However,  the value of extrinsic parameter is much different. I think that the problem is the camera intrinsic parameter. I want to ask  you how to calibrate your camera intrinsic parameter (I use Matlab)?  Or, maybe I deploy myself lab environment is not good? 
PS: The attached image which show the color between the pointcloud and image are not matched. 

Thank you, bless you have the good day.

中文版本
你好
我想請教一下問題
就是我將光達(vlp-16)與相機(realsense)像你那樣擺設
然後開始進行校正,但我校正出來的值差很多
我推測有可能是相機內參問題
因此想請問你相機內參是如何校正的呢？
目前我是使用matlab
亦或是我的實驗環境擺設有問題呢？

PS:底下可以很明顯看到我的點雲上image的顏色沒對齊
[image]
[image]

感謝您, 祝您有個美好的一天

ChongjianYUAN, 2021-05-21:
Hello, We use OpenCV to calibrate the camera's internal parameters, but Matlab should be able to achieve the same effect. From the first picture we can see that the point cloud is dense, not like the point cloud of VLP16, is it a preprocessed point cloud?

yanlong658, 2021-05-21:
Yes, I use the lidar-IMU odometry to recover the point cloud. It is like as your Fast-lio. Because of the point cloud is sparse, I use the lio system to recover the point cloud.  However, the lidar and camera ocllusion problem are serios.

中文版
是的,我有預先處理點雲,使用lio系統,該系統跟你們實驗室的Fast-lio差不多.
因為vlp-16點雲太少,所以使用lio系統恢復點雲,
但又另外會產生嚴重的zero-valued and multi-valued.

ChongjianYUAN, 2021-05-21:
> Yes, I use the lidar-IMU odometry to recover the point cloud. It is like as your Fast-lio. Because of the point cloud is sparse, I use the lio system to recover the point cloud. However, the lidar and camera occlusion problem are serious.
> 
> 中文版
> 是的,我有預先處理點雲,使用lio系統,該系統跟你們實驗室的Fast-lio差不多.
> 因為vlp-16點雲太少,所以使用lio系統恢復點雲,
> 但又另外會產生嚴重的zero-valued and multi-valued.

Firstly, make sure the camera's internal parameters are accurate.
Secondly, Try both configuration modes separately. If the edge extraction result is not good(e.g. too few edges), adjust the Edge.min_dis_threshold, Edge.max_dis_threshold to 0.05 and 0.1. 

Edge.min_dis_threshold: 0.03Edge.max_dis_threshold: 0.06

ChongjianYUAN, 2021-05-21:
Maybe you can wait for me to release the next version to support the Velodyne-camera calibration. The expected time is next week.

zzkslam, 2021-07-09:
> Velodyne-camera calibration

Can you tell me when will support the Velodyne-camera calibration.Thanks in advance

ChongjianYUAN, 2021-07-12:
> > Velodyne-camera calibration
> 
> Can you tell me when will support the Velodyne-camera calibration.Thanks in advance

The earliest is the middle of August. Because I've been busy with other projects. Thank you for your understanding

yanlong658, 2021-08-02:
English version
Sorry, I still have some question.
I use your code to test my scene.
The image and point cloud are match perfectly on rviz.
However, the computed extrinsic parameter is not good.
I think that the edge is not evenly.

I currently  solve the rotation before translation.
What do you think?

Thank you.

中文版
你好,我還是有些小問題想請教
我持續使用的的code測試我的環境
在rviz上面點雲跟image都匹配的很好
但是計算出來的外部參數差滿多的
目前是推測試edge分散不均勻
但測了三個場所都是類似情況
目前是先計算旋轉在計算平移
想請問您有什嘛看法嘛？
[image]
[image]
感謝您

ChongjianYUAN, 2021-08-02:
谢谢你的测试，单个场景标定的外参差距是比较大的，因为单一场景的约束数量比较少，如果希望获得一个比较统一的外参，你可以等待后续的联合标定的功能（多场景标定同一外参），经过我们的验证，联合标定能够得到一个比较一致的外参，但是release时间会是在八月中下旬，当然你也可以尝试自己修改代码先实现这个功能，谢谢！

ChongjianYUAN, 2021-08-02:
> English version
> Sorry, I still have some question.
> I use your code to test my scene.
> The image and point cloud are match perfectly on rviz.
> However, the computed extrinsic parameter is not good.
> I [...]

看上去雷达边缘偏少，请问是点云累计时间不够吗

yanlong658, 2021-08-03:
English version
Hi, sir

My point cloud accumulates for about 30 seconds by vlp-16.
The number of distant objects (>50m) is indeed relatively small than close object.
I have also tried to adjust the voxel size parameter.
However, if the adjustment is too small, the edge will have a small segment of the edge.

Thank you

中文版

你好
我點雲累積大概30秒左右
採用的是vlp-16
個人是覺的
較遠的物體點(>50m)雲數量確實比較少
但較近的物體(<30m)點雲數量算是滿密集的
也有試著調過voxel大小參數
但調太小的話,edge會有滿多一小段一小段的edge

感謝您
[image]

ChongjianYUAN, 2021-08-12:
> English version
> Hi, sir
>
> My point cloud accumulates for about 30 seconds by vlp-16.
> The number of distant objects (>50m) is indeed relatively small than close object.
> I have also tried to adjust the [...]

Sorry for the late reply. Judging from the quality of your point cloud, the mapping accuracy of your point cloud is relatively low. My parameters are set based on the avia static point cloud, so it may not work well on your dataset. I will solve the problem of Velodyne calibration at the end of August. Thank you for your patience and understanding.

hr2894235132, 2022-10-24:
> Firstly, make sure the camera's internal parameters are accurate. Secondly, Try both configuration modes separately. If the edge extraction result is not good(e.g. too few edges), adjust the [...]

您好！想请问一下，这两个参数的意义分别是什么呢？这将有助于我进行调参。

zhuoyuan0, 2023-01-26:
Hi thanks for your excellent work.
I run the code both with indoor and outdoor data but the result is not good like the demo shows.  Could you please tell me how to choose the scenes for the algorithm? Indoor and outdoor, which one will be better? And Is there any other keys I should pay attention to? Thanks very much.