hku-mars/livox_camera_calib #9: 错误的结果

nkliuhui, 2021-05-25:
我下载了源码并成功编译安装, 数据集使用readme中链接提供的;运行室内1和室外1, 都没有得到好的对齐结果;
根据初值, 二者的匹配效果正常; rough calibration阶段把结果拉偏了, 优化的结果更差; 其中室内数据的匹配结果如下:
运行命令为 roslaunch livox_camera_calib calib_indoor.launch
[image]

nkliuhui, 2021-05-25:
请问是可能哪里出了问题

ChongjianYUAN, 2021-05-26:
感谢你的测试，我更新了edge extraction的参数让rough calibration可以更加准确，在本地测试已解决这个问题

nkliuhui, 2021-05-27:
不好意思哈, 今天测试了一下, 发现初值没有前天测试的时候给的好; 在粗配准阶段是激光与图像先靠近, 然后突然向反方向跳变了; 后期优化阶段拉回来一点, 但最终结果也不是太理想. 请问是和计算平台的依赖库有关吗? Eigen 3.3.7, Ubuntu 16, Ceres 2.0.0, PCL 1.7.
[image]

[image]

ChongjianYUAN, 2021-05-27:
> 不好意思哈, 今天测试了一下, 发现初值没有前天测试的时候给的好; 在粗配准阶段是激光与图像先靠近, 然后突然向反方向跳变了; 后期优化阶段拉回来一点, 但最终结果也不是太理想. 请问是和计算平台的依赖库有关吗? Eigen 3.3.7, Ubuntu 16, Ceres 2.0.0, PCL 1.7.
> [image]
> 
> [image]

请问有使用最新版本的代码吗，或者说有修改过哪些参数吗

nkliuhui, 2021-05-27:
我重新编译了代码, 发现还是同样的结果; 确实使用了最新的代码, 我怀疑可能是我安装环境的问题, 后续换台电脑再尝试一下哈.
(在粗配准阶段, 开始时确实是朝着重合的方向收敛, 但是突然跳变了.)
[image]
[image]

nkliuhui, 2021-05-27:
没有修改参数文件, 只修改了数据包路径

ChongjianYUAN, 2021-05-27:
好的，感谢你的辛苦测试，如果初值提供的较好的话，可以在源代码中注释掉add_error（为初值增加误差）和roughCalib（粗配准）进行标定

CanCanZeng, 2021-07-30:
我也是只改了bag包的路径，然后运行程序，标定的结果也不好。我的环境是ubuntu16.04, eigen3.3.4, pcl-1.7, ceres-1.14.0。你找到问题的原因了吗?   @nkliuhui

CanCanZeng, 2021-07-30:
我测试了indoor1.bag 和outdoor1.bag，都是类似的情况Rough calibration刚开始几次是朝着正确的方向在优化，后来就突然朝着错误的方向一直走了。  试过了把add_error去掉，也不对。最后去掉了roughCalib得到的结果看起来才是正确的，不过也还是存在一些明显的匹配错误，例如outdoor1.bag柱子上面颜色贴错了，可能是rgb图内参有点误差造成的。

[image]

请问这是正常现象吗，我感觉使用的是提供的demo数据，应该是不做任何修改就能得到正确的结果才对。 @ChongjianYUAN

ChongjianYUAN, 2021-07-31:
> 我测试了indoor1.bag 和outdoor1.bag，都是类似的情况Rough calibration刚开始几次是朝着正确的方向在优化，后来就突然朝着错误的方向一直走了。 试过了把add_error去掉，也不对。最后去掉了roughCalib得到的结果看起来才是正确的，不过也还是存在一些明显的匹配错误，例如outdoor1.bag柱子上面颜色贴错了，可能是rgb图内参有点误差造成的。
> 
> [image]
> 
> 请问这是正常现象吗，我感觉使用的是提供的demo数据，应该是不做任何修改就能得到正确的结果才对。 @ChongjianYUAN

最后的上色点云是正常的现象，因为积累的雷达点云通常会比原始物体大一些（因为雷达发射的射线是一束光斑而不是理想的射线），加上内参的影响，时会出现在边缘处一些地方颜色贴错的现象，至于roughCalib, 请问是在最新的代码上测试的吗

nkliuhui, 2021-07-31:
> 我也是只改了bag包的路径，然后运行程序，标定的结果也不好。我的环境是ubuntu16.04, eigen3.3.4, pcl-1.7, ceres-1.14.0。你找到问题的原因了吗? @nkliuhui

我放弃了

CanCanZeng, 2021-07-31:
是的，我就是在最新的代码上测试的 @ChongjianYUAN

ChongjianYUAN, 2021-10-20:
Please try with the latest code, thanks. Earlier versions were susceptible to RANSAC randomness