hku-mars/livox_camera_calib #10: work with slam

duanqg, 2021-05-27:
Your work is great and inspiring. 
I want to put this code together with my SLAM program. Do you think it is feasible?
 If possible, I would like to know the specific interface
 thank you

ChongjianYUAN, 2021-05-31:
Good advice. I think the calibration process can be used to detect any misalignment of the sensors during their online operation. Besides, it can provide precise extrinsic parameters at initialization time. The key is the synchronization of data and time between sensors. I will think about the detailed implementation later.

duanqg, 2021-05-31:
thank you!
 I am reading the Fastlio of your lab now, which is also very inspiring and very consistent with your code, because Fastlio is also based on avia feature points for fusion, and I have achieved good robustness after partial algorithm modification. It would be icing on the cake if we could combine the depth camera for feature recognition and tight coupling!
[image]