hku-mars/livox_camera_calib #14: 机械式LℹDAR rosbag录制咨询

jhx646018057, 2021-06-18:
你好，我用的是速腾128线机械LiDAR，想采用此方案进行和相机的联合标定，这样需要的rosbag大概录制多久的？我理解livox是非重复扫描的可能需要录制的rosbag长一点。而机械式LiDAR是重复扫描的，那我这机械LiDAR 录制10帧是不是就可以了？

ChongjianYUAN, 2021-06-18:
> 你好，我用的是速腾128线机械LiDAR，想采用此方案进行和相机的联合标定，这样需要的rosbag大概录制多久的？我理解livox是非重复扫描的可能需要录制的rosbag长一点。而机械式LiDAR是重复扫描的，那我这机械LiDAR 录制10帧是不是就可以了？

我之前使用过64线的ouster雷达进行测试时，发现此类雷达在静止时，由于重复扫描，所以帧数对结果的影响不大，但是在静止条件下提取的边缘很有限，而且不够精准，所以建议先使用雷达里程计（Loam or Lio）配准几帧雷达数据，让点云更加稠密，然后再使用该方案进行相机和雷达的联合标定，会效果好很多，下面是ouster雷达的一个案例：
[image]
使用配准后的稠密点云能够很高精度的标定外参：
[image]

jhx646018057, 2021-06-21:
好的，谢谢你的回复

closed 2021-06-21 by jhx646018057