hku-mars/livox_camera_calib #19: pcl problem

Zhu-Liyuan, 2021-07-22:
Hi, thank you for your contribution to the society.
When I run your code, I met a problem:

[ INFO] [1626940489.756662588]: Loading the rosbag /home/catkin_ws/src/livox_camera_calib/data/merge.bag
[ INFO] [1626940490.902931052]: Sucessfully load Point Cloud and Image
[ INFO] [1626940490.903897494]: Building Voxel
[ INFO] [1626940508.755300261]: Extracting Lidar Edge
[ INFO] [1626940526.485055384]: Load all data!
[ INFO] [1626940526.904265229]: Finish prepare!
[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!

Do you know in which function this error happens? So I can check my own data.
Thank you for your time.

ChongjianYUAN, 2021-07-22:
> Hi, thank you for your contribution to the society.
> When I run your code, I met a problem:
>
> [ INFO] [1626940489.756662588]: Loading the rosbag /home/catkin_ws/src/livox_camera_calib/data/merge.bag
> [ [...]

It seems that you failed to specify the correct point cloud topic name in the config.yaml file. Then the point cloud is empty. Try to use "rosbag info merge.bag" to get the correct topic name. Besides, if the LiDAR point cloud type is custom_msg, you should change "Data.custom_msg" to 1 in config.yaml.

Zhu-Liyuan, 2021-07-27:
Thank you for your reply. 
I did changed the config file to satify my own data and the topic name is correct.
But the RVIZ could load my point cloud and generate planners.
[image]
The terminal is like this:
[image]

ChongjianYUAN, 2021-07-27:
> Thank you for your reply.
> I did changed the config file to satify my own data and the topic name is correct.
> But the RVIZ could load my point cloud and generate planners.
> [image]
> The terminal is like this:
> [image]

Could you share this bag file to me (ycj1@connect.hku.hk)? Then I will try to debug and reply to you. Thanks

ChongjianYUAN, 2021-08-04:
It seems that the way your sensor is installed is different from mine, which will lead to no projection points if you use the default initial extrinsic.  Could you send an image to show your sensor suit with coordinates?

closed 2021-11-25 by ChongjianYUAN

sorbicacid, 2025-06-01:
I encountered a similar issue. The problem occurred when I was modifying the configuration file. The parameter matrix should be separated by commas, not spaces. Since the parameter matrix I copied was separated by spaces, it prompted the error mentioned above. After changing it to commas, the calculation worked normally.