hku-mars/livox_camera_calib #20: Bad result of calibration

meng-zha, 2021-08-02:
Lidar: Livox Mid-100
Camera: Flir BlackFly
Rosbag: https://cloud.tsinghua.edu.cn/f/09ceccd8e1c647f59650/?dl=1
Camera_config: https://cloud.tsinghua.edu.cn/f/2687d771b0ea421ea30d/?dl=1

The calibration result seems promising after rough calibration, by collapsed after fine-tune optimization.
[image]
[image]

ChongjianYUAN, 2021-08-12:
Sorry for my late reply. As you can see from the LiDAR edge extraction results, this is not a good calibration scene because there are a lot of bleeding points and the edge information is not obvious. I recommend trying to collect data in an open outdoor structure scene.

rahulsharma11, 2021-08-19:
Hi, i am able to calibrate the livox and camera with this repo.

Also got some output results-
[image]

My question is where to use these values? I am not sure any calibration matrix is there.... Does r2live requires this output ?

Thanks.

ChongjianYUAN, 2021-08-19:
> Hi, i am able to calibrate the livox and camera with this repo.
> 
> Also got some output results-
> [image]
> 
> My question is where to use these values? I am not sure any calibration matrix is there.... Does r2live requires this output ?
> 
> Thanks.

Hi, the final calibration matrix can be found in result/extrinsic.txt

rahulsharma11, 2021-08-19:
Hi, thanks for the reply.

I got 3x4 extrinsic matrix with few more output like-

0.00184784,-0.999967,-0.00792853,0.115226
-0.0028004,0.00792334,-0.999965,-0.0171396
0.999994,0.00186998,-0.00278566,0.0134214
0,0,0,1
179.894,-179.84,179.546,0,0,0

So to use in r2live calib file, please correct me if i am wrong-
**extrinsic rotation -** 
0.00184784,-0.999967,-0.00792853
-0.0028004,0.00792334,-0.999965
0.999994,0.00186998,-0.00278566

**extrinsic translation -** 
0.115226
-0.0171396
0.0134214

Is this correct?

And Where should i use "179.894,-179.84,179.546,0,0,0"?

ChongjianYUAN, 2021-08-19:
You are right. The last row "179.894,-179.84,179.546,0,0,0" is the Euler Angle for the nominal division and is saved just for debug. The final matrix is extrinsic from the LiDAR frame to the Camera frame. If the extrinsic in calib file in r2live is also defined from the LiDAR frame to camera frame, you can use it directly. Otherwise, you should use, rotation = r^(-1),  translation = -r^(-1) * t, where r, t is the origin rotation and translation in result/extrinsic.txt.

rahulsharma11, 2021-08-19:
Thank you very much.

closed 2021-10-07 by ChongjianYUAN

hr2894235132, 2022-10-24:
> 

您好，我在标定时遇到了和您一样的问题，想问一下，您最后是通过什么得到了解决呢？