hku-mars/livox_camera_calib #21: Issue while running on ros-kinetics based docker image on ubuntu machine18.04

rahulsharma11, 2021-08-17:
Hi,
Because of lower version of PCL(1.17), i prepared a docker image(ros:kinetic-ros-core-xenial), with all dependencies. Also placed the respective files on path.
While running i am getting the following error-

**[ INFO] [1629189514.066569998]: Loading the rosbag /opt/catkin_ws/src/livox_camera_calib/data/indoor1.bag**
process[rviz-3]: started with pid [381]
process 381: D-Bus library appears to be incorrectly set up; failed to read machine uuid: UUID file '/etc/machine-id' should contain a hex string of length 32, not length 0, with no other text
See the manual page for dbus-uuidgen to correct this issue.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
**[rviz-3] process has died [pid 381, exit code -11, cmd /opt/ros/kinetic/lib/rviz/rviz -d /opt/catkin_ws/src/livox_camera_calib/rviz_cfg/calib.rviz __name:=rviz __log:=/root/.ros/log/81017cba-ff36-11eb-bf90-2c56dc38a7a2/rviz-3.log].
log file: /root/.ros/log/81017cba-ff36-11eb-bf90-2c56dc38a7a2/rviz-3*.log**
[ INFO] [1629189515.516145283]: Sucessfully load Point Cloud and Image
[ INFO] [1629189515.516546482]: Building Voxel
[ INFO] [1629189539.422697927]: Extracting Lidar Edge
[ERROR] [1629189544.520860218]: Can not load image from /opt/catkin_ws/src/livox_camera_calib/data/indoor1.bag
**[lidar_camera_calib-2] process has died [pid 374, exit code 255, cmd /opt/catkin_ws/devel/lib/livox_camera_calib/lidar_camera_calib /opt/catkin_ws/src/livox_camera_calib/config/camera.yaml /opt/catkin_ws/src/livox_camera_calib/config/config_outdoor.yaml /opt/catkin_ws/src/livox_camera_calib/data/indoor1.bag /opt/catkin_ws/src/livox_camera_calib/result __name:=lidar_camera_calib __log:=/root/.ros/log/81017cba-ff36-11eb-bf90-2c56dc38a7a2/lidar_camera_calib-2.log].
log file: /root/.ros/log/81017cba-ff36-11eb-bf90-2c56dc38a7a2/lidar_camera_calib-2*.log**

I am not sure, why process has been died and telling indoor1.bag is not found but it has already started with the same.

Any suggestions ?
Thanks.

rahulsharma11, 2021-08-18:
Its working now.

closed 2021-08-18 by rahulsharma11

Xulei-Zhai, 2021-10-11:
Did you change anything? I have the same problem as you.

zaalsabb, 2022-06-17:
@rahulsharma11 Hi, can you please share your dockerfile? Thanks!