hku-mars/livox_camera_calib #22: 运行时错误

Dorothy-2016, 2021-08-23:
您好，首先感谢您的工作让Liar-Cam标定更便捷！因最近也在做这方面的工作，所以试了一下您的方法，遇到一个运行问题，如下图：
[image]
程序总是在这个位置直接报错，也没有其他提示，可能是什么原因呢？

ChongjianYUAN, 2021-08-24:
你好，请问你的ros和pcl版本是多少？

Dorothy-2016, 2021-08-24:
> 你好，请问你的ros和pcl版本是多少？

ros是kinetic版， pcl1.7版

ChongjianYUAN, 2021-08-24:
请问rviz有正常输出吗？

Dorothy-2016, 2021-08-25:
> 请问rviz有正常输出吗？

rviz也没有正常输出
[image]

ChongjianYUAN, 2021-08-25:
> > 请问rviz有正常输出吗？
> 
> rviz也没有正常输出
> [image]
> 我单独运行rosbag play outdoor1.bag文件是有图像输出的

这个情况看上去像是没有成功读取点云，这个outdoor1.bag是从我给的数据链接下载的demo数据吗

ChongjianYUAN, 2021-08-29:
> > > > 请问rviz有正常输出吗？
> > > 
> > > 
> > > rviz也没有正常输出
> > > [image]
> > > 我单独运行rosbag play outdoor1.bag文件是有图像输出的
> > 
> > 
> > 这个情况看上去像是没有成功读取点云，这个outdoor1.bag是从我给的数据链接下载的demo数据吗
> 
> 嗯嗯，是的，google云盘上下载的那个压缩包，我单独play bag文件的时候也是只看到图像话题数据，没有看到点云数据，开始我以为是tf没给对所以没显示

可以用rosbag info outdoor1.bag再次查看下是否有点云数据吗？

Dorothy-2016, 2021-08-30:
> > > > > 请问rviz有正常输出吗？
> > > > 
> > > > 
> > > > rviz也没有正常输出
> > > > [image]
> > > > 我单独运行rosbag play outdoor1.bag文件是有图像输出的
> > > 
> > > 
> > > 这个情况看上去像是没有成功读取点云，这个outdoor1.bag是从我给的数据链接下载的demo数据吗
> > 
> > 
> > 嗯嗯，是的，google云盘上下载的那个压缩包，我单独play bag文件的时候也是只看到图像话题数据，没有看到点云数据，开始我以为是tf没给对所以没显示
> 
> 可以用rosbag info outdoor1.bag再次查看下是否有点云数据吗？
[image]
[image]
[image]
这是查看的结果，好像确实没有Lidar数据？

Dorothy-2016, 2021-08-30:
[image]
Finally, I saw the Lidar data when I set the fixed frame as livox_frame when run rviz on the outdoor1.bag. So it can be sure that the data has no problem!
but when I run roslaunch livox_camera_calib calib.launch, it still output error like the follow picture:
[image]

ChongjianYUAN, 2021-09-03:
> [image]
> Finally, I saw the Lidar data when I set the fixed frame as livox_frame when run rviz on the outdoor1.bag. So it can be sure that the data has no problem!
> but when I run roslaunch livox_camera_calib calib.launch, it still output error like the follow picture:
> [image]

你好，可以在launch文件中加入 launch-prefix="gdb -ex run --args" 打开gdb模式看看具体是在哪个地方报错的

xiaobai-1-1, 2021-11-24:
I met the same question, is there a solution？

ChongjianYUAN, 2021-11-25:
> I met the same question, is there a solution？

Is it tested on the latest code?

xiaobai-1-1, 2021-11-25:
I used the code  downloaded today, but it's the same problem.
testdata is outdoor2.bag.

ChongjianYUAN, 2021-11-25:
> I used the code downloaded today, but it's the same problem. testdata is outdoor2.bag.

Could you provide the ros version and pcl version

xiaobai-1-1, 2021-11-25:
ubuntu 16.04 ros-kinetic  pcl-1.7

chenhongjin811, 2022-01-06:
我也遇到了同样的问题，ubuntu 16.04, kinetic, pcl-1.8.1

wangmengjiao12, 2022-01-11:
请问，一样的问题，有解决方法了吗？

987428377, 2022-08-30:
我也遇到了相同的问题，我在尝试用自己录制的bag包，请问以下环境是静止的吗？我并没有找到作者的outdoor2.bag压缩包瞎子阿地址

Skyhfz, 2023-06-08:
I met the same issue. I use ROS GDB to detect the fxxking problem. The result points to lidar-calib_camera.hpp line 771. I review this code and find no bugs. Finally, The reason is that we have 2 PCL versions. 
we can use the specific version by find_packadge(exact) to solve this problem