hku-mars/livox_camera_calib #25: os1 64 and hik vision camera caliberation.

islamtalha01, 2021-10-11:
Hi
I have checked your git repository regarding lidar camera calibration. I am a research intern at my Home university NUST Pakistan. I have an os1 64 spinning lidar available at my lab. I need to calibrate the camera and lidar. Now kindly tell me the precise step I need to follow to get that calibration parameter. I don't know how to use Fast LIO for spinning lidar calibration. kindly outline the steps to follow to get extrinsic parameters with your package using os1 64 lidar. I would very thankful to you for this favor.I have bag file recorded with topic pointcloud2 and hivision camera image. kindly help me in this regard

AFEICHINA, 2021-10-13:
> can you tell me when you will be releasing your version for spinning lidar-like ouster os1 64?. i need that one. if you have anyone kindly do provide me. otherwise, tell me how to preprocess my ouster lidar data so it can be calibrated. kindly help in this regard. you also used ouster in your video so kindly help me.

I create a repo with floam and livox_camera_calibration, you can test it by modifing config files and some codes for os1 64. Fast-Lio can get better PCD map, but you should calib your LiDAR and external imu first. It will introducing more works. In my test, floam also can get fine calibration results  in indoor enviroment.
https://github.com/AFEICHINA/expend_lidar_camera_calib.git

islamtalha01, 2021-10-13:
Ok can you tell me which file and specific code of lines I need to change
for this purpose? the second thing I have recorded rosbag with point cloud
and image topic so will foam can be applied on the rosbag file and its
point cloud become dense? or it needs to run on sensor stream and then I
have to record again with dense cloud.

closed 2021-11-25 by ChongjianYUAN

aniket-gupta1, 2022-09-29:
@islamtalha01 Were you able to calibrate your ouster lidar with your camera?