hku-mars/livox_camera_calib #29: Question about vpnp cost function

RoseWillow, 2021-11-12:
Dear author, thanks for your excellent work, but I am puzzled by the vpnp cost function, I did not understand why we need the line direction to compute the matrix V, and the physical meaning of this formula V*R*V.transpose

[image]

Look forward to your reply

ChongjianYUAN, 2021-11-16:
> Dear author, thanks for your excellent work, but I am puzzled by the vpnp cost function, I did not understand why we need the line direction to compute the matrix V, and the physical meaning of this formula V_R_V.transpose
> 
> [image]
> 
> Look forward to your reply

[image]
the line direction can help us build point-to-line residual but not point-to-point residual. Besides, the correct formulation of the point-to-line residual is shown in this picture. There are some minor bugs in the code, but they don't matter much after testing.

RoseWillow, 2021-11-16:
Thank you for your reply, but I still don't know why you use this formula 
[image]
 to build point-to-line residuals and the physical meaning of this formula。I think we should the formula in your paper 
[image]

ChongjianYUAN, 2021-11-25:
> Thank you for your reply, but I still don't know why you use this formula [image] to build point-to-line residuals and the physical meaning of this formula。I think we should use the formula in your paper [image]
> 
> What's more, I know the vector ni in the code is the direction of the line while it represent the normal vector in your paper。I do not understand why the single residual ri is a two-dimensional vector in your last reply formula.

Two-dimensional vector means the pixel difference in two dimensions. Also, you can use one dimensional residual, which means the distance in the image plane.

closed 2022-01-09 by ChongjianYUAN

tdcsu, 2024-05-16:
> Thank you for your reply, but I still don't know why you use this formula [image] to build point-to-line residuals and the physical meaning of this formula。I think we should use the formula in your [...]

Hi, have you understanded this block of the code, or the physical meaning of the formula V_R?
best
TD

tdcsu, 2024-05-16:
> Hi, have you understanded this block of the code, or the physical meaning of the formula V_R? best TD

Now it is clear.  please see the draft below：
[image]
L_a is a lidar edge line in pixel cordinate. Point a is a lidar edge point after the undist-pinhole-tansform form lidar frame. Its corresponding image edge point b in image pixle line L_b;
it is clear that: a=pi(exp(^)L^P_i);
b=p_i;
so in the code R=(I-V)R_1 =I*R_1-V*R_1
R_1=a-b;
V=n_i.trans*n_i, ni means the direction vector of L_b
the V is a projection matrix, a vector left multiply the V therefore a vector, here is the vector <b,c> in the draft.
so here it is! 
R=(I-V)R_1 =I*R_1-V*R_1=vector<b,a>-vector<b,c>=vector<c,a>,which indicate the 3d-2d residual

note: the discription above cover the noisy situation. if no noisy, the vector<b,a> is the residual, no noisy situation is also covered in the darft when vector<b,c>=0