hku-mars/livox_camera_calib #30: 标定参数平移部分相差较多

Elder-Dai, 2021-11-18:
你好，请教一个问题，标定的参数中，旋转部分比较准确，但平移部分相差了大概10几厘米，请问是场景选的太远了吗？由于平移部分不准，远处的点对齐的比较好，距离5米以内的点效果比较差

ChongjianYUAN, 2021-11-19:
> 你好，请教一个问题，标定的参数中，旋转部分比较准确，但平移部分相差了大概10几厘米，请问是场景选的太远了吗？由于平移部分不准，远处的点对齐的比较好，距离5米以内的点效果比较差

[image]
是的，可以参照这个雅可比矩阵，当场景里的边缘都距离较远时(Z_i比较大)，平移参数的雅可比会比较小，从而标定出来平移误差比较大

Elder-Dai, 2021-11-22:
> > 你好，请教一个问题，标定的参数中，旋转部分比较准确，但平移部分相差了大概10几厘米，请问是场景选的太远了吗？由于平移部分不准，远处的点对齐的比较好，距离5米以内的点效果比较差
> 
> [image] 是的，可以参照这个雅可比矩阵，当场景里的边缘都距离较远时(Z_i比较大)，平移参数的雅可比会比较小，从而标定出来平移误差比较大

好的，非常感谢，我选了一个比较近的场景，效果好多了

closed 2021-11-22 by samsdolphin