hku-mars/livox_camera_calib #33: 运行自己的数据时错误

ltf1412, 2021-12-12:
您好，我在运行下载的indoor和outdoor的bag都没有问题，但是运行自己的indoor数据时出现报错，我把相机的内参和image的topic都改了，路径也改过了，出现以下报错：
[image]
在rviz中的画面是这样的：
[image]
请问这是什么问题呢？

ChongjianYUAN, 2021-12-12:
> 您好，我在运行下载的indoor和outdoor的bag都没有问题，但是运行自己的indoor数据时出现报错，我把相机的内参和image的topic都改了，路径也改过了，出现以下报错： [image] 在rviz中的画面是这样的： [image] 请问这是什么问题呢？ 我用的ubuntu18.04 melodic和pcl1.8

您好，请问可以把这个出错的数据分享给我一份吗，我的邮箱 ycj1@connect.hku.hk

ChongjianYUAN, 2021-12-13:
> 您好，我在运行下载的indoor和outdoor的bag都没有问题，但是运行自己的indoor数据时出现报错，我把相机的内参和image的topic都改了，路径也改过了，出现以下报错： [image] 在rviz中的画面是这样的： [image] 请问这是什么问题呢？ 我用的ubuntu18.04 melodic和pcl1.8

您好，请检查内参矩阵和Camera.width, Camera.height是否输入正确，
[image]
从terminal输出的结果看，这个内参矩阵是错误的

ltf1412, 2021-12-13:
您好，我按照正确的格式改了相机矩阵，依然报错

ChongjianYUAN, 2021-12-14:
> 您好，我按照正确的格式改了相机矩阵，依然报错
> […](#)

Camera.width和Camera.height参数修改了吗

ltf1412, 2021-12-18:
感谢您之前的帮助，现在我可以成功运行了，但是得到的结果很差，激光雷达和图片视野相差太多，图片为2048x1536像素

                           

我用的是lidar mid 70和azure kinect3相机，请问这是因为mid70的FOV是圆形导致的吗？

ChongjianYUAN, 2021-12-23:
请问，是否是在安装时没有考虑到Fov的重叠呢，你可以根据mid70的fov和相机的fov大概画一下fov的重叠范围，如果fov重叠很少的话，我的方法是不适用的，需要借助传感器运动来进行标定

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

ltf1412, 2021-12-30:
您好，相机和激光雷达硬件安装示意图如下：

录制indoor和outdoor数据时相机和雷达的中间视野重叠范围也是很多的

                  

                      

相机拍摄RGB时的模式设置如下，选择了接近圆形视场的4:3纵横比，分辨率为2048×1536

但是最终结果相差很大：

 
 
 
 
 
 &amp;nbsp; &amp;nbsp; &amp;nbsp; &amp;nbsp; &amp;nbsp; 
 
 
 
 
 
 
 
 ------------------&amp;nbsp;原始邮件&amp;nbsp;------------------ 
 主题:&amp;nbsp;Re: [hku-mars/livox_camera_calib] 运行自己的数据时错误 (Issue #33)

ChongjianYUAN, 2022-01-01:
> 您好，相机和激光雷达硬件安装示意图如下： 录制indoor和outdoor数据时相机和雷达的中间视野重叠范围也是很多的                         相机拍摄RGB时的模式设置如下，选择了接近圆形视场的4:3纵横比，分辨率为2048×1536 但是最终结果相差很大：          
> […](#)
> ------------------ 原始邮件 ------------------ 发件人: "hku-mars/livox_camera_calib" ***@***.***&gt;; 发送时间: 2021年12月23日(星期四) 下午3:41 ***@***.***&gt;; ***@***.******@***.***&gt;; 主题: Re: [hku-mars/livox_camera_calib] 运行自己的数据时错误 (Issue #33) 请问，是否是在安装时没有考虑到Fov的重叠呢，你可以根据mid70的fov和相机的fov大概画一下fov的重叠范围，如果fov重叠很少的话，我的方法是不适用的，需要借助传感器运动来进行标定 从 Windows 版邮件<[https://go.microsoft.com/fwlink/?LinkId=550986&gt;发送](https://go.microsoft.com/fwlink/?LinkId=550986&gt;%E5%8F%91%E9%80%81)
> ________________________________ 发件人: ltf1412 ***@***.***&gt; 发送时间: Saturday, December 18, 2021 9:59:41 PM 收件人: hku-mars/livox_camera_calib ***@***.***&gt; 抄送: ycj ***@***.***&gt;; Comment ***@***.***&gt; 主题: Re: [hku-mars/livox_camera_calib] 运行自己的数据时错误 (Issue #33) 感谢您之前的帮助，现在我可以成功运行了，但是得到的结果很差，激光雷达和图片视野相差太多，图片为2048x1536像素 &amp;nbsp; &amp;nbsp; &amp;nbsp; &amp;nbsp; &amp;nbsp; 我用的是lidar mid 70和azure kinect3相机，请问这是因为mid70的FOV是圆形导致的吗？

不好意思，图片好像挂掉了，能重新编辑一下吗

ltf1412, 2022-01-02:
您好，图片以附件形式发送。
相机和激光雷达硬件安装示意图如图1所示：

录制indoor和outdoor数据时相机和雷达的中间视野重叠范围也是很多的，室内和室外的录制图如图2、图3所示

                  

                      

相机拍摄RGB时的模式设置如下，选择了接近圆形视场的4:3纵横比，分辨率为2048×1536，工作模式如图4所示

但是最终结果相差很大，如图5、图6所示

ChongjianYUAN, 2022-01-09:
你好，我并没有在附件找到图片，请问可以重新发送一下吗，可以发送到 ***@***.******@***.***> 邮箱，谢谢

从 Windows 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>发送

Chatoyant19, 2023-06-01:
你好，请问你的“激光雷达和图片视野相差太多”这个问题解决了吗，谢谢