hku-mars/livox_camera_calib #36: 运行自己数据时无法收敛

ailiwei17, 2022-01-08:
我不知道如何录制覆盖图片所有位置的点云，是这种原因导致了错误吗？
或许您知道如何不改变雷达位置录制覆盖图片所有位置的点云吗
[image]
[image]
[image]

ChongjianYUAN, 2022-01-09:
> 我不知道如何录制覆盖图片所有位置的点云，是这种原因导致了错误吗？ 或许您知道如何不改变雷达位置录制覆盖图片所有位置的点云吗 [image] [image] [image]

请问是什么激光雷达呢，fov好像特别小

ailiwei17, 2022-01-13:
livox mid-40

cyn-liu, 2022-09-28:
请问使用这种范围很小的livox激光雷达，如何录制覆盖图片所有位置的点云？

ailiwei17, 2022-09-28:
> 请问使用这种范围很小的livox激光雷达，如何录制覆盖图片所有位置的点云？

可以在比较大的场景标定

closed 2022-09-28 by ailiwei17

cyn-liu, 2022-09-29:
> > 请问使用这种范围很小的livox激光雷达，如何录制覆盖图片所有位置的点云？
> 
> 可以在比较大的场景标定

你好，我没有理解你的意思，我面临的问题是激光雷达获得的点云视场范围小于图片的视场。