hku-mars/livox_camera_calib #40: Issue with rectified image input

aditdoshi333, 2022-03-08:
Hello,

I tried with non rectified images and rectified images. Calibration output is much better with non rectified images. Is the code is expecting distorted images? Are you running rectification in the pipeline anywhere?

Regards

ChongjianYUAN, 2022-03-08:
> Hello,
> 
> I tried with non rectified images and rectified images. Calibration output is much better with non rectified images. Is the code is expecting distorted images? Are you running rectification in the pipeline anywhere?
> 
> Regards

Yes, the input image should be distorted, and the point cloud projection needs to be provided with the same distortion model

aditdoshi333, 2022-03-08:
Okay!

But does the final calibration matrix need to use with the rectified image or non rectified image?

ChongjianYUAN, 2022-03-08:
> Okay!
> 
> But does the final calibration matrix need to use with the rectified image or non rectified image?

It depends on whether you correlate the LiDAR and camera data through projection, and if so, you need to use the rectified image.

aditdoshi333, 2022-03-08:
I have one more doubt. I am able to calibrate pretty well with your code but not able to use the same matrix in r3live.  From the following can you say how is calibration?
[image]

ChongjianYUAN, 2022-03-15:
> I have one more doubt. I am able to calibrate pretty well with your code but not able to use the same matrix in r3live. From the following can you say how is calibration? [image]

Sorry for the late reply. The extrinsic matrix in livox_camera_calib is from LiDAR coordinate to Camera coordinate and in R3LIVE is from Camera coordinate to LiDAR coordinate. Therefore, you should use the inverse of the output result of livox_camera_calib.