hku-mars/livox_camera_calib #41: 数据准备

huangsiyuan0717, 2022-03-15:
感谢您的工作。
我准备录取horizon + realsense的数据，我想把他们都录取成一段简短的bag包，然后转换成相应的pcd和图片，因为帧率不同，如何确认他们是同一帧呢。

HungryBen, 2022-03-19:
你好，请问如何将一段时间的雷达数据（bag文件），积累点云数据成一个pad文件呢，期待回复！

ChongjianYUAN, 2022-03-19:
> 你好，请问如何将一段时间的雷达数据（bag文件），积累点云数据成一个pad文件呢，期待回复！

你好，使用launch文件李的bag_to_pcd.launch就可以讲bag文件转成pcd文件，注意修改数据路径和bag文件包中雷达数据的topic名称

huangsiyuan0717, 2022-03-23:
@ChongjianYUAN 感谢回复，录制了标定数据，但是效果似乎不是很好，想问问可能是哪些参数的问题
[image]
[image]

ChongjianYUAN, 2022-03-23:
> @ChongjianYUAN 感谢回复，录制了标定数据，但是效果似乎不是很好，想问问可能是哪些参数的问题 [image] [image]

请问这两个是标定场景吗，看上去场景的边缘比较少，可以尝试几组数据放在一起使用multi_calib标定

closed 2022-03-31 by huangsiyuan0717