hku-mars/livox_camera_calib #42: pcl problem

ailiwei17, 2022-03-17:
#19 
您好，物品遇到了同样的问题，已经把Data.custom_msg: 1改好了
[image]
感觉点云已经加载进去了但还是会报错
[image]

ChongjianYUAN, 2022-03-18:
> #19 您好，我遇到了同样的问题，已经把Data.custom_msg: 1改好了，并确认了路径 [image] 感觉点云已经加载进去了但还是会报错 [image]

看上去雷达提取的edge并没有成功投影到相机平面上，可以看一下initial extrinsic对应的投影图像吗，另外这个室内场景选择的并不好，有效的边缘很少

ailiwei17, 2022-03-19:
> > #19 您好，我遇到了同样的问题，已经把Data.custom_msg: 1改好了，并确认了路径 [image] 感觉点云已经加载进去了但还是会报错 [image]
> 
> 看上去雷达提取的edge并没有成功投影到相机平面上，可以看一下initial extrinsic对应的投影图像吗，另外这个室内场景选择的并不好，有效的边缘很少
initial [extrinsic对应的投影图像指的是这一幅图吗？好像确实没有点云的样子，我的初始的外参矩阵是使用[https://github.com/Livox-SDK/livox_camera_lidar_calibration](url)得到的，投影验证时发现不够准确，但应该不会偏差太大
[image]

HungryBen, 2022-03-19:
您好，您的点云数据是如何积累的呢？

ChongjianYUAN, 2022-03-19:
> 您好，您的点云数据是如何积累的呢？

我是直接使用livox官方给的[livox_ros_driver](https://github.com/Livox-SDK/livox_ros_driver)录制了一段静止的数据

ChongjianYUAN, 2022-03-19:
> > > #19 您好，我遇到了同样的问题，已经把Data.custom_msg: 1改好了，并确认了路径 [image] 感觉点云已经加载进去了但还是会报错 [image]
> > 
> > 
> > 看上去雷达提取的edge并没有成功投影到相机平面上，可以看一下initial extrinsic对应的投影图像吗，另外这个室内场景选择的并不好，有效的边缘很少
> 
> initial extrinsic对应的投影图像指的是这一幅图吗？好像确实没有点云的样子，我的初始的外参矩阵是使用[https://github.com/Livox-SDK/livox_camera_lidar_calibration](url)得到的，投影验证时发现不够准确，但应该不会偏差太大。后来我也修改为默认的外参矩阵，依然没有点云 [image]

请再确认下相机参数是否修改成你的设备的参数

ailiwei17, 2022-03-19:
> 请再确认下相机参数是否修改成你的设备的参数

不好意思，是我疏忽了，似乎是因为原厂给的相机的内参矩阵只是1080P的，我拍照设置为720P，导致了错误。谢谢！

closed 2022-03-19 by ailiwei17