hku-mars/livox_camera_calib #43: 运行multi_calib出现错误

BIT-wangzb, 2022-03-24:
在运行multi_calib是出现这种错误，请问是什么原因？是图像边缘特征太少吗
[image]
[image]
？

ChongjianYUAN, 2022-03-24:
> 在运行multi_calib是出现这种错误，请问是什么原因？是图像边缘特征太少吗？ [image] [image]

请问rviz里输出的图像是怎样的呢

ChongjianYUAN, 2022-03-25:
> > > 在运行multi_calib是出现这种错误，请问是什么原因？是图像边缘特征太少吗？ [image] [image]
> > 
> > 
> > 请问rviz里输出的图像是怎样的呢
> > [image]
> > 是这样的，激光雷达成功提取了特征，图像没有提取成功。

这看起来雷达数据中没有提取到任何有效的边缘呀

zhangyuran-gg, 2024-10-15:
> > > > 在运行multi_calib是出现这种错误，请问是什么原因？是图像边缘特征太少吗？ [image] [image]
> > > 
> > > 
> > > 请问rviz里输出的图像是怎样的呢
> > > [image]
> > > 是这样的，激光雷达成功提取了特征，图像没有提取成功。
> 
> 这看起来雷达数据中没有提取到任何有效的边缘呀

Hello! The effect of using calibration bars and grid plates is not very good. If I want to obtain better data, which scenarios are recommended to use?