hku-mars/livox_camera_calib #44: 雷达和相机的初始偏差过大导致标定效果不佳

fl840249479, 2022-04-07:
换边缘信息更丰富的场景也是这样，因为初始偏差过大会选择就近的边缘拟合没法出正确的结果
[image]
[image]

ChongjianYUAN, 2022-04-07:
> 换边缘信息更丰富的场景也是这样，因为初始偏差过大会选择就近的边缘拟合没法出正确的结果 [image] [image]

可以考虑使用multi_calib，几个场景的数据一起优化会对初值更加鲁棒一些；此外，这个场景对于雷达点云来说，边缘信息并不丰富，因为我们主要提取的是深度连续的边缘

fl840249479, 2022-04-10:
> > 换边缘信息更丰富的场景也是这样，因为初始偏差过大会选择就近的边缘拟合没法出正确的结果 [image] [image]
> 
> 可以考虑使用multi_calib，几个场景的数据一起优化会对初值更加鲁棒一些；此外，这个场景对于雷达点云来说，边缘信息并不丰富，因为我们主要提取的是深度连续的边缘
感谢您的回答。这个问题解决了，错误原因是我使用了错误的相机内参。我一开始使用了相机自带的程序输出了相机内参，这个内参与相机的实际内参存在较大偏差，导致程序中雷达和相机初始偏差过大没法拟合在一起

closed 2022-04-10 by fl840249479

ly-uuu, 2022-04-12:
> 换边缘信息更丰富的场景也是这样，因为初始偏差过大会选择就近的边缘拟合没法出正确的结果 [image] [image]
同学，可以加个联系方式吗，关于标定和R3LIVE想和你讨论下。QQ：790629782

Tomato1107, 2022-04-13:
> 换边缘信息更丰富的场景也是这样，因为初始偏差过大会选择就近的边缘拟合没法出正确的结果 [image] [image]

其实可以尝试手动调整下相机安装角度或朝向，使相机和lidar之间的初始偏差偏小一些