hku-mars/livox_camera_calib #46: calcLine的具体计算方法原理是什么？

chengih, 2022-04-25:
[image]
为什么要计算
matrix[1][4] = a1 * x1 + b1 * y1 + c1 * z1;

ChongjianYUAN, 2022-04-26:
> [image] 为什么要计算 matrix[1][4] = a1 * x1 + b1 * y1 + c1 * z1;
> 
> [image] calc(matrix, point); calc中的计算原理是什么？
> 
> 以及后续分别计算6次得到的point各代表的含义是什么？

你好，calcLine目的是计算两个平面在voxel内部的交线. 实际上平面的交线是无限长的，我们只关心在voxel内部的交线；所以可以通过判断两个平面与voxel的六个面的交点，把在voxel内的交点连线，即是平面相交的线。calc函数求解的是一个行列式，求解的point是输入的三个平面的交点。

chengih, 2022-04-26:
感谢回复，
那matrix[1][4] = a1 * x1 + b1 * y1 + c1 * z1;
代表什么含义了？

chengih, 2022-04-27:
明白了，感谢回复

closed 2022-05-11 by ChongjianYUAN