hku-mars/livox_camera_calib #48: Rough Optimization looks good, then Optimization goes bad

pauldeee, 2022-05-07:
Hi there,

Cool package, hoping to get this to work! I'm using an os1-64. When I launch the calib the Rough Optimization starts and is pretty good. However, when it moves onto the Optimization step, the point cloud is aligned very poorly. It seems when Optimization starts the cloud is rotated by some amount, maybe 90 degrees.

Here are a couple photos. Please note that my camera images do not fit on my monitor so I can only show what I can see:

Rough Optimization step:
[image]

Optimization Step:
[image]

Any thoughts or ideas would be appreciated. Thank you

ChongjianYUAN, 2022-05-07:
> Hi there,
>
> Cool package, hoping to get this to work! I'm using an os1-64. When I launch the calib the Rough Optimization starts and is pretty good. However, when it moves onto the Optimization step, [...]

> Hi there,
>
> Cool package, hoping to get this to work! I'm using an os1-64. When I launch the calib the Rough Optimization starts and is pretty good. However, when it moves onto the Optimization step, [...]

Is it the wrong projection from the beginning of optimization, or during optimization?

pauldeee, 2022-05-07:
> Is it the wrong projection from the beginning of optimization, or during optimization?

It is from the beginning of the optimization step.

ChongjianYUAN, 2022-05-07:
> > Is it the wrong projection from the beginning of optimization, or during optimization?
> 
> It is from the beginning of the optimization step.

could you share me the dataset by email ycj1@connect.hku.hk

closed 2022-05-10 by ChongjianYUAN

sunwell1994, 2022-06-16:
I have the same issue. The rough optimization stage looks better but the optimization stage makes the alignment drift away.

seajayshore, 2022-09-01:
For some setups we have a similar issue - rough optimisation works nicely, then fine optimisation drifts away after a while.

Did you find the solution to your version of this problem? Can you maybe share publicly if so?

sunwell1994, 2022-09-01:
> For some setups we have a similar issue - rough optimisation works nicely, then fine optimisation drifts away after a while.
> 
> Did you find the solution to your version of this problem? Can you maybe share publicly if so?

If you just need to perform one-time calibration,  you can modify the library to save all the intermediate extrinsic matrics and then calibrate the rotation and translation manually.