hku-mars/livox_camera_calib #49: 启动时遇到错误

FengYuQ, 2022-05-08:
[image]
启动程序的时候[lidar_camera_calib-2]进程被杀死，想请问下这是什么原因导致的呢，有什么解决方法嘛？
我的平台是jeston TX2, ubuntu18.04, Ros melodic, PCL 1.8.1

FengYuQ, 2022-05-09:
looking forward to your early reply

ChongjianYUAN, 2022-05-09:
> [image] 启动程序的时候[lidar_camera_calib-2]进程被杀死，想请问下这是什么原因导致的呢，有什么解决方法嘛？ 我的平台是jeston TX2, ubuntu18.04, Ros melodic, PCL 1.8.1

你好，看样子是没能成功读取配置文件，可能是修改yaml文件后格式出了问题，请检查一下报错是否出现在loadCalibConfig函数中

FengYuQ, 2022-05-09:
你好，我中午在calib.launch文件中添加了launch -prefix="gdb -ex run --args"来启动gdb进行错误查看，结果如下图所示。
想请问下参数文件具体需要什么格式呢？有什么更好的办法定位错误嘛？

FengYuQ, 2022-05-09:
你好，我中午在calib.launch文件中添加了launch -prefix="gdb -ex run --args"来启动gdb进行错误查看，结果如下图所示。
想请问下参数文件具体需要什么格式呢？有什么更好的办法定位错误嘛？ 
[image]

ChongjianYUAN, 2022-05-10:
> 你好，我中午在calib.launch文件中添加了launch -prefix="gdb -ex run --args"来启动gdb进行错误查看，结果如下图所示。 想请问下参数文件具体需要什么格式呢？有什么更好的办法定位错误嘛？ [image]

你好，这个repo目前在18.04上跑会出错，我还在定位原因

FengYuQ, 2022-05-10:
> > 你好，我中午在calib.launch文件中添加了launch -prefix="gdb -ex run --args"来启动gdb进行错误查看，结果如下图所示。 想请问下参数文件具体需要什么格式呢？有什么更好的办法定位错误嘛？ [image]
> 
> 你好，这个repo目前在18.04上跑会出错，我还在定位原因

我目前定位到出错的位置是在lidar_camera_calib.cpp中203行的
`Calibration calibra(image_file, pcd_file, calib_config_file);`
calibra函数出问题，但具体原因仍未找到，如果您找到原因的话麻烦告诉我一下

FengYuQ, 2022-05-12:
你好，请问有什么进展嘛

ChongjianYUAN, 2022-05-14:
> 你好，请问有什么进展嘛

你好，最近我正在忙于修改文章，预计还需要一周左右的时间才有空维护以前的repo，谢谢理解

shiyang24, 2022-06-06:
> 你好，请问有什么进展嘛

可能是opencv版本问题，将cmakelists里改成find_package(OpenCV 3 REQUIRED)试下

cherie6866, 2022-06-06:
你好,你的邮件我已经收到.我会尽快查看并回复.

XinquanDu, 2022-07-21:
你好，我也遇到了一样的问题：
[image]
经过排查之后发现是改过之后的config_outdoor.yaml在calibration函数里面没有打开成功。
配置是ubuntu18.04+pcl1.8.1+openCV 3.2.0

Freeman-Cai, 2023-04-21:
> > 你好，我中午在calib.launch文件中添加了launch -prefix="gdb -ex run --args"来启动gdb进行错误查看，结果如下图所示。 想请问下参数文件具体需要什么格式呢？有什么更好的办法定位错误嘛？ [image]
> 
> 你好，这个repo目前在18.04上跑会出错，我还在定位原因
你好，请问这个问题现在解决了吗

Freeman-Cai, 2023-04-21:
> > 你好，我中午在calib.launch文件中添加了launch -prefix="gdb -ex run --args"来启动gdb进行错误查看，结果如下图所示。 想请问下参数文件具体需要什么格式呢？有什么更好的办法定位错误嘛？ [image]
> 
> 你好，这个repo目前在18.04上跑会出错，我还在定位原因

你好，请问这个问题现在得到解决了吗

yahyahajlaoui, 2023-07-27:
hello, is the issues resolved ?