hku-mars/livox_camera_calib #52: 雷达点云稀疏无法提取足够的边缘特征

SylaraAnh, 2022-05-28:
今天尝试在仿真环境中进行标定，出现了以下的错误：
OpenCV Error: Assertion failed (npoints >= 0 && (depth == CV_32F || depth == CV_64F)) in projectPoints, file /home/wjc/ws_opencv/opencv-3.4.16/modules/calib3d/src/calibration.cpp, line 3317
[image]
貌似是提取出来的边缘特征太少了，请问标定时对雷达或者拍摄的视角有什么建议吗。我仿真环境中雷达的点云比较稀疏。可以通过调节配置文件中的参数得到收敛的结果吗

SylaraAnh, 2022-05-28:
debug了一下好像是太稀疏了没提取出来任何边缘特征

ChongjianYUAN, 2022-05-28:
你好，点云太稀疏了是无法提取边缘的，如果是重复扫描雷达(velodyne, hesai这种)，可以考虑跑个LO或者LIO算法把点云积累的稠密一些再进行投影，如果是非重复扫描雷达(avia, horizon等)，可以用更长的采集时间得到稠密点云

fireinth, 2023-05-06:
> 你好，点云太稀疏了是无法提取边缘的，如果是重复扫描雷达(velodyne, hesai这种)，可以考虑跑个LO或者LIO算法把点云积累的稠密一些再进行投影，如果是非重复扫描雷达(avia, horizon等)，可以用更长的采集时间得到稠密点云

您好，我使用非重复扫描雷达（livox hap）进行积累点云 ，但发现噪点特别多，您的avia在积累时有做去噪方面的处理吗

dingkwang, 2023-08-17:
@ChongjianYUAN 一般需要多稠密的点云呢？如何知道是否提取到边缘？

Richard-Cold, 2024-07-29:
请问，问题解决了吗？是因为点云太稀疏的问题吗？