hku-mars/livox_camera_calib #54: Empty argument `lidar_line_cloud_3d` passed to buildVPnp() causing a cv::Exception. 是因为点云密度太小没有提取到有效边缘吗？

groundturtle, 2022-06-01:
你好，我在运行程序的时候opencv抛出异常：
```bash
  what():  OpenCV(4.2.0) ../modules/core/src/matrix_expressions.cpp:23: error: (-5:Bad argument) Matrix operand is an empty matrix. in function 'checkOperandsExist'
```
经过检查，错误发生在`buildVPnp()`（或者`buildPnP()`，当我把`bool use_vpnp`改为`false`。
在这个函数中，由于第五个参数`lidar_line_cloud_3d`的size为0，没有点能够进入`std::vector<cv::Point3f> pts_3d`，在进行投影时发生了错误。
```c++
// src/include/lidar_camera_calib.hpp, 
void Calibration::buildVPnp(
    const Vector6d &extrinsic_params, const int dis_threshold,
    const bool show_residual,
    const pcl::PointCloud<pcl::PointXYZ>::Ptr &cam_edge_cloud_2d,
    const pcl::PointCloud<pcl::PointXYZI>::Ptr &lidar_line_cloud_3d,
    std::vector<VPnPData> &pnp_list)
{
  //....
  // project 3d-points into image view
  std::vector<cv::Point2d> pts_2d;
  // debug
  std::cout << "camera_matrix:" << camera_matrix << std::endl;
  std::cout << "distortion_coeff:" << distortion_coeff << std::endl;
  std::cout << "r_vec:" << r_vec << std::endl;
  std::cout << "t_vec:" << t_vec << std::endl;
  // here output 0 to screen.
  std::cout << "pts 3d size:" << pts_3d.size() << std::endl;
  // cv::Exception thrown here.
  cv::projectPoints(pts_3d, r_vec, t_vec, camera_matrix, distortion_coeff,
                    pts_2d);
  //...
```
我没有足够的知识获得更多信息。请问这种情况是否因为我的点云密度太小或场景内没有足够的物体边缘而发生？我能否通过增加扫描时间（我使用非重复式扫描激光雷达livox）来避免这个问题?

另外，ceres提示我许多接口已经被放弃或即将在2.2.0中被移除，全部都是警告，没有发生错误。
```bash
cat /usr/local/include/ceres/version.h
```
```c
#define CERES_VERSION_MAJOR 2
#define CERES_VERSION_MINOR 2
#define CERES_VERSION_REVISION 0
```
这里显示我使用的ceres版本为2.2.0，但`LocalParameterization()`并没有如warning信息所说的被移除。请问这些API变化是否会影响标定功能？

ChongjianYUAN, 2022-06-16:
> 

API变换不会影响标定功能，如果是点云密度太小，可以通过增加扫描时间来避免这个问题，livox系列雷达建议扫描时间在20s左右，可以得到一个相对稠密的雷达点云

groundturtle, 2023-04-13:
> > 你好，我在运行程序的时候opencv抛出异常：
> > ```shell
> >   what():  OpenCV(4.2.0) ../modules/core/src/matrix_expressions.cpp:23: error: (-5:Bad argument) Matrix operand is an empty matrix. in function 'checkOperandsExist'
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 经过检查，错误发生在`buildVPnp()`（或者`buildPnP()`，当我把`bool use_vpnp`改为`false`。 在这个函数中，由于第五个参数`lidar_line_cloud_3d`的size为0，没有点能够进入`std::vector<cv::Point3f> pts_3d`，在进行投影时发生了错误。
> > ```c++
> > // src/include/lidar_camera_calib.hpp, 
> > void Calibration::buildVPnp(
> >     const Vector6d &extrinsic_params, const int dis_threshold,
> >     const bool show_residual,
> >     const pcl::PointCloud<pcl::PointXYZ>::Ptr &cam_edge_cloud_2d,
> >     const pcl::PointCloud<pcl::PointXYZI>::Ptr &lidar_line_cloud_3d,
> >     std::vector<VPnPData> &pnp_list)
> > {
> >   //....
> >   // project 3d-points into image view
> >   std::vector<cv::Point2d> pts_2d;
> >   // debug
> >   std::cout << "camera_matrix:" << camera_matrix << std::endl;
> >   std::cout << "distortion_coeff:" << distortion_coeff << std::endl;
> >   std::cout << "r_vec:" << r_vec << std::endl;
> >   std::cout << "t_vec:" << t_vec << std::endl;
> >   // here output 0 to screen.
> >   std::cout << "pts 3d size:" << pts_3d.size() << std::endl;
> >   // cv::Exception thrown here.
> >   cv::projectPoints(pts_3d, r_vec, t_vec, camera_matrix, distortion_coeff,
> >                     pts_2d);
> >   //...
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 我没有足够的知识获得更多信息。请问这种情况是否因为我的点云密度太小或场景内没有足够的物体边缘而发生？我能否通过增加扫描时间（我使用非重复式扫描激光雷达livox）来避免这个问题?
> > 另外，ceres提示我许多接口已经被放弃或即将在2.2.0中被移除，全部都是警告，没有发生错误。
> > ```shell
> > cat /usr/local/include/ceres/version.h
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > ```c
> > #define CERES_VERSION_MAJOR 2
> > #define CERES_VERSION_MINOR 2
> > #define CERES_VERSION_REVISION 0
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 这里显示我使用的ceres版本为2.2.0，但`LocalParameterization()`并没有如warning信息所说的被移除。请问这些API变化是否会影响标定功能？
> 
> 你好，你有解决么？我也遇到了这个问题

解决了，我这里就是点云密度不够的问题，但那些弃用的接口我也不知道有什么影响。不过我反复标定的效果都很差。

czy1998916, 2024-07-22:
@groundturtle 请问如何增加点云密度，我用了fastlio建图，也出现了这种问题
[image]
[image]