hku-mars/livox_camera_calib #57: 运行多场景标定时出错

EDISION654, 2022-06-30:
非常感谢作者所做的工作！我在运行roslaunch multi_calib.launch时，终端报以下错误。
数据集是从README中提供的网址下载的，路径也改成了本地路径，请问作者出现这种问题是什么原因呢？
[image]

ChongjianYUAN, 2022-07-03:
> 非常感谢作者所做的工作！我在运行roslaunch multi_calib.launch时，终端报以下错误。 数据集是从README中提供的网址下载的，路径也改成了本地路径，请问作者出现这种问题是什么原因呢？ [image]

你好，确定路径是 “/0.bmp” 吗

EDISION654, 2022-07-04:
> > 非常感谢作者所做的工作！我在运行roslaunch multi_calib.launch时，终端报以下错误。 数据集是从README中提供的网址下载的，路径也改成了本地路径，请问作者出现这种问题是什么原因呢？ [image]
> 
> 你好，确定路径是 “/0.bmp” 吗

你好！是的，我将下载的图像数据包（包含0.bmp、1.bmp、2.bmp）放在了自己的路径/sd/system/image下；将雷达数据包（包含0.pcd、1.pcd、2.pcd）放在了路径/sd/system/pcd2下。

ChongjianYUAN, 2022-07-04:
> > > 非常感谢作者所做的工作！我在运行roslaunch multi_calib.launch时，终端报以下错误。 数据集是从README中提供的网址下载的，路径也改成了本地路径，请问作者出现这种问题是什么原因呢？ [image]
> > 
> > 
> > 你好，确定路径是 “/0.bmp” 吗
> 
> 你好！是的，我将下载的图像数据包（包含0.bmp、1.bmp、2.bmp）放在了自己的路径/sd/system/image下；将雷达数据包（包含0.pcd、1.pcd、2.pcd）放在了路径/sd/system/pcd2下。

可以用 ls /0.bmp 命令看看是否有输出

987428377, 2022-08-30:
我认为，多场景的化你要给的是地址的/sd/system/image，而不是/sd/system/image/0.bmp，