hku-mars/livox_camera_calib #59: 运行自己的数据错误。

chenxinyuzdshuai, 2022-08-02:
非常感谢作者所做的工作！我使用自己的数据集，我的传感器为livox-AVIA和灰点相机BFLY-PSG-23S63C。安装方式和论文中的一样。运行环境为ubuntu18.04，ros melodic ,pcl-1.8。在运行roslaunch multi_calib.launch时，终端报以下错误。
/home/xinyuc/图片/2022-08-02 17-55-46屏幕截图.png

chenxinyuzdshuai, 2022-08-02:
[image]

ChongjianYUAN, 2022-08-04:
> [image]

请检查相机和雷达的安装方式，修改初始的旋转矩阵

chenxinyuzdshuai, 2022-08-09:
谢谢，已经检查和修改。

ChongjianYUAN, 2022-08-11:
> 谢谢，已经检查和修改。

请问修改后还是报错吗

Clarence-Feng, 2022-09-22:
感谢作者的工作！请问我按照上述安装方式，确保了初始旋转矩阵没问题后还是会报同样的错误是为什么呢；是否跟相机拍摄图片的清晰度有关，谢谢！

ChongjianYUAN, 2022-09-23:
> 

有可能的，如果相机图像中不能提取出边缘（或者雷达点云不能提取边缘）的话，就会报同样的错误

Goughhhhhh, 2023-08-10:
> 

我发现是和影像的分辨率有关，我最开始采集的影像分辨率是640×480，就出现和题主同样的错误，后来改成1920×1080就好了。请问要如何使用640×480的影像作为标定数据呢？