hku-mars/livox_camera_calib #61: Multi Calibration not Converging/gets stuck

ckgpeace, 2022-08-25:
Hi Team

Thanks for the awesome repository.

I was able to run Single Extrinsic Calibration but when I run multi calibration, it just gets stuck forever. Multi Calibration does not even work for 2 image and 2 pcd files. I am attaching my snapshot of yaml file I used for multi calibration. I am also attaching the output after running multi calibration. I have used same same multi calibration file without any change except the input yaml file.

Please help me out if I am missing something. 

Regards

[image]
[image]
.

ckgpeace, 2022-09-01:
This is resolved using the previous commit

Mastergalen, 2022-09-01:
Looks like a debug `getchar()` was left behind in the latest commit 55b1423259b3d459e8621a0b2be849cc6e5ae132.

If you delete line 352 in `lidar_camera_multi_calib.cpp` it will fix the issue.

https://github.com/hku-mars/livox_camera_calib/blob/master/src/lidar_camera_multi_calib.cpp#L352

```
      Eigen::Quaterniond opt_q(R);
      std::cout << "q_dis:" << RAD2DEG(opt_q.angularDistance(q))
                << " ,t_dis:" << (T - ori_t).norm() << std::endl;
-     getchar();
      // if (opt_q.angularDistance(q) < DEG2RAD(0.01) &&
      //     (T - ori_t).norm() < 0.005) {
      //   opt_flag = false;
      // }
      // if (!opt_flag) {
      //   break;
      // }
```

ChongjianYUAN, 2022-09-23:
> Looks like a debug `getchar()` was left behind in the latest commit [55b1423](https://github.com/hku-mars/livox_camera_calib/commit/55b1423259b3d459e8621a0b2be849cc6e5ae132).
> 
> If you delete line 352 in `lidar_camera_multi_calib.cpp` it will fix the issue.
> 
> https://github.com/hku-mars/livox_camera_calib/blob/master/src/lidar_camera_multi_calib.cpp#L352
> 
> ```
>       Eigen::Quaterniond opt_q(R);
>       std::cout << "q_dis:" << RAD2DEG(opt_q.angularDistance(q))
>                 << " ,t_dis:" << (T - ori_t).norm() << std::endl;
> -     getchar();
>       // if (opt_q.angularDistance(q) < DEG2RAD(0.01) &&
>       //     (T - ori_t).norm() < 0.005) {
>       //   opt_flag = false;
>       // }
>       // if (!opt_flag) {
>       //   break;
>       // }
> ```

Thanks for the reminder.

closed 2022-09-23 by ChongjianYUAN