hku-mars/livox_camera_calib #62: Code understanding in vpnp_calib

BayRanger, 2022-08-26:
Hi there, 

In the file lidar_camera_multi_calib.cpp， I get stuck to understand the following code in the vpnp_calib, which is for ceres optimization.
`      residuals[0] = ud - T(pd.u);
      residuals[1] = vd - T(pd.v);
      Eigen::Matrix<T, 2, 2> I =
          Eigen::Matrix<float, 2, 2>::Identity().cast<T>();
      Eigen::Matrix<T, 2, 1> n = pd.direction.cast<T>();
      Eigen::Matrix<T, 1, 2> nt = pd.direction.transpose().cast<T>();
      Eigen::Matrix<T, 2, 2> V = n * nt;
      V = I - V;
      V = I - V;
      Eigen::Matrix<T, 2, 2> R = Eigen::Matrix<float, 2, 2>::Zero().cast<T>();
      R.coeffRef(0, 0) = residuals[0];
      R.coeffRef(1, 1) = residuals[1];
      R = V * R * V.transpose();
      residuals[0] = R.coeffRef(0, 0);
      residuals[1] = R.coeffRef(1, 1);
`
could you please explain the motivation to do this calculation,?  especially about R and V.f

Bests,
BR

autumn999999, 2023-10-19:
Hi！Did you understand the meaning of this code? I also have the same question. I’d be grateful if you could help me.

tdcsu, 2024-05-15:
> Hi！Did you understand the meaning of this code? I also have the same question. I’d be grateful if you could help me.

please see issue #29