hku-mars/livox_camera_calib #65: 关于数据包录制格式的问题

hr2894235132, 2022-10-13:
您好！非常感谢您优秀的工作。我在使用时遇到了一些问题。
想请问一下，我在录制雷达点云时，录制标准pcl::pointcloud2还是自定义的Custom_msg呢？不知道对结果会不会有什么影响？（我使用的是livox mid 70雷达）

ChongjianYUAN, 2022-10-16:
> 您好！非常感谢您优秀的工作。我在使用时遇到了一些问题。 想请问一下，我在录制雷达点云时，录制标准pcl::pointcloud2还是自定义的Custom_msg呢？不知道对结果会不会有什么影响？（我使用的是livox mid 70雷达）

都可以，没有影响，在config文件里设置是否是custom_msg就可以

hr2894235132, 2022-10-16:
> > 您好！非常感谢您优秀的工作。我在使用时遇到了一些问题。 想请问一下，我在录制雷达点云时，录制标准pcl::pointcloud2还是自定义的Custom_msg呢？不知道对结果会不会有什么影响？（我使用的是livox mid 70雷达）
> 
> 都可以，没有影响，在config文件里设置是否是custom_msg就可以

好的非常感谢！！！

hr2894235132, 2022-10-17:
[image]
您好，0代表custom_msg,1代表pointcloud2是吧？
还有在Indoor配置文件中并没有这一参数，是不需要嘛？是否在室外效果更好？

ChongjianYUAN, 2022-10-17:
> [image] 您好，0代表custom_msg,1代表pointcloud2是吧？ 还有在Indoor配置文件中并没有这一参数，是不需要嘛？是否在室外效果更好？

您好，1代表custom_msg，0代表pointcloud2. Indoor配置文件应该是忘记加了，不好意思，默认用得是pointcloud2, 这个参数对结果没有任何影响，唯一的区别是custom msg是livox的msg，pointcloud2是ros标准的点云msg。

hr2894235132, 2022-10-18:
> > [image] 您好，0代表custom_msg,1代表pointcloud2是吧？ 还有在Indoor配置文件中并没有这一参数，是不需要嘛？是否在室外效果更好？
> 
> 您好，1代表custom_msg，0代表pointcloud2. Indoor配置文件应该是忘记加了，不好意思，默认用得是pointcloud2, 这个参数对结果没有任何影响，唯一的区别是custom msg是livox的msg，pointcloud2是ros标准的点云msg。

好的，非常感谢您的耐心回答！

hr2894235132, 2022-10-24:
> > > [image] 您好，0代表custom_msg,1代表pointcloud2是吧？ 还有在Indoor配置文件中并没有这一参数，是不需要嘛？是否在室外效果更好？
> > 
> > 
> > 您好，1代表custom_msg，0代表pointcloud2. Indoor配置文件应该是忘记加了，不好意思，默认用得是pointcloud2, 这个参数对结果没有任何影响，唯一的区别是custom msg是livox的msg，pointcloud2是ros标准的点云msg。
> 
> 好的，非常感谢您的耐心回答！

您好！我在后续使用mid-70测试的时候遇到了一些问题。
**1.点云plane和edge提取的问题：**
下面是mid-70采集到的点云信息，相机的图片信息，以及在计算时残差的图。感觉提取到的边太过于少，是否需要更改一下配置文件中体素分辨率以及平面拟合的一些参数？
[image]
[image]
[image]

**2.优化时会朝着过拟合的方向一直进行，导致结果很不好，我猜想还是因为提取到的edge太少的问题，可以帮我解答一下嘛？万分感谢！！！困扰太久了。**
[image]

ChongjianYUAN, 2022-10-25:
> 您好！我在后续使用mid-70测试的时候遇到了一些问题。 **1.点云plane和edge提取的问题：** 下面是mid-70采集到的点云信息，相机的图片信息，以及在计算时残差的图。感觉提取到的边太过于少，是否需要更改一下配置文件中体素分辨率以及平面拟合的一些参数？ [image] [image] [image] [...]

边缘过少的话，建议使用multi_calib，多采集几个场景（至少3个），标定会更鲁棒和准确一些

hr2894235132, 2022-10-25:
> 边缘过少的话，建议使用multi_calib，多采集几个场景（至少3个），标定会更鲁棒和准确一些

您好！非常感谢您在百忙中恢复，我就是使用的multi_calib，一共采集了三个场景如下：
[image]
[image]
[image]
**我猜想因为雷达edge提取的问题，影响了最后的标定精度结果，效果并不是很好**（如下图所示）
[image]
**可以给我一点调参建议或者提高特征提取的效果的方法吗？还是需要再添加更多的场景去进行标定呢？**

ChongjianYUAN, 2022-10-25:
> 您好！非常感谢您在百忙中恢复，我就是使用的multi_calib，一共采集了三个场景如下： [image] [image] [image] **我猜想因为雷达edge提取的问题，影响了最后的标定精度结果，效果并不是很好**（如下图所示） [image] **可以给我一点调参建议或者提高特征提取的效果的方法吗？还是需要再添加更多的场景去进行标定呢？**

稍微增大config文件中的
Edge.min_dis_threshold: 0.03->0.4
Edge.max_dis_threshold: 0.06->0.8
可以提取到更多没那么准确的边缘，另外，MID70本身FOV比较小，建议多添加场景，或者先用lio或者livox_loam算法先把点云聚集起来，当成一帧pcd使用

hr2894235132, 2022-10-26:
> 稍微增大config文件中的 Edge.min_dis_threshold: 0.03->0.4 Edge.max_dis_threshold: 0.06->0.8 可以提取到更多没那么准确的边缘，另外，MID70本身FOV比较小，建议多添加场景，或者先用lio或者livox_loam算法先把点云聚集起来，当成一帧pcd使用

您好，非常感谢您的建议！我按照您的建议，调整config文件中的参数，Edge.min_dis_threshold：0.03->0.05  Edge.max_dis_threshold: 0.06->0.1 Plane.normal_theta_min: 30->60 Plane.normal_theta_max: 150->120  ，特征edge提取变多，最终得到了比较理想的结果。
[image]
[image]

最后再次感谢您的帮助！祝您生活愉快，工作顺利！

tangyubbb, 2022-12-20:
> 您好！我在后续使用mid-70测试的时候遇到了一些问题。 **1.点云plane和edge提取的问题：** 下面是mid-70采集到的点云信息，相机的图片信息，以及在计算时残差的图。感觉提取到的边太过于少，是否需要更改一下配置文件中体素分辨率以及平面拟合的一些参数？ [image] [image] [image] [...]

麻烦问一下楼主收集点云数据的方法是什么呀。我只会在bag里面抽取一帧一帧的pcd数据。谢谢！（因为mid70没有imu，你是用外接imu，然后用lio之类的方法得到的pcd吗？）

tangyubbb, 2022-12-20:
> 麻烦问一下楼主收集点云数据的方法是什么呀。我只会在bag里面抽取一帧一帧的pcd数据。谢谢！（因为mid70没有imu，你是用外接imu，然后用lio之类的方法得到的pcd吗？）

tangyubbb, 2022-12-20:
我现在把他们加在一起了，

hr2894235132, 2022-12-21:
> 麻烦问一下楼主收集点云数据的方法是什么呀。我只会在bag里面抽取一帧一帧的pcd数据。谢谢！（因为mid70没有imu，你是用外接imu，然后用lio之类的方法得到的pcd吗？）

imu我用的相机自带的，bag转成pcd工程里有对应的包，直接运行就可以了

tangyubbb, 2022-12-21:
> imu我用的相机自带的，bag转成pcd工程里有对应的包，直接运行就可以了

我才注意到这个，非常感谢你的回答！参照你们以上的问答，也为我的工作提供了很大的帮助。

closed 2022-12-29 by hr2894235132