hku-mars/livox_camera_calib #66: 论文中的实验数据

FengYuQ, 2022-10-13:
您好，关于论文中的实验数据有个问题想请教一下您，关于鲁棒性评估中，Fig13左边那张关于RPY的图是如何计算的，我用您提供的数据尝试了几个场景都没有达到这么小的角度差异。
[image]

ChongjianYUAN, 2022-10-16:
> 您好，关于论文中的实验数据有个问题想请教一下您，关于鲁棒性评估中，Fig13左边那张关于RPY的图是如何计算的，我用您提供的数据尝试了几个场景都没有达到这么小的角度差异。 [image]

你好，这张图的RPY是用外参移除了标称值后计算的，此外，该版本代码经过多次bug修复的commit，结果可能略微有所变动

FengYuQ, 2022-10-17:
> > 您好，关于论文中的实验数据有个问题想请教一下您，关于鲁棒性评估中，Fig13左边那张关于RPY的图是如何计算的，我用您提供的数据尝试了几个场景都没有达到这么小的角度差异。 [image]
> 
> 你好，这张图的RPY是用外参移除了标称值后计算的，此外，该版本代码经过多次bug修复的commit，结果可能略微有所变动

好滴，感谢您的回复！