hku-mars/livox_camera_calib #68: seg.segment 出现 std::bad_alloc 的错误

hhz334, 2022-11-01:
您好，我的设备是ubuntu20.04 pcl 1.10 
ros中——catkin_make没有问题
在运行roslaunch _camera_calib calib.launch 时，出现了terminate called after throwing an instance of 'std::bad_alloc'  what():  std::bad_alloc错误
麻烦您帮忙看看，问题可能出在哪里？
[image]

ChongjianYUAN, 2022-11-15:
> 您好，我的设备是ubuntu20.04 pcl 1.10 ros中——catkin_make没有问题 在运行roslaunch _camera_calib calib.launch 时，出现了terminate called after throwing an instance of 'std::bad_alloc' what(): std::bad_alloc错误 麻烦您帮忙看看，问题可能出在哪里？ [image]

你好，使用的是自己采集的数据集吗

WeiZixu-HIT, 2024-07-22:
请问解决了吗