hku-mars/livox_camera_calib #70: Depth discontinuous boundary problem

Xiaotianzhi, 2022-11-14:
Thank you very much for your excellent work. We are experimenting with your method, but we have encountered some problems. I noticed that you compared the depth continuous boundary with the depth discontinuous boundary in the paper and provided a direct image (Figure 7 in the paper). Could you please help to provide the engineering code for depth discontinuous boundary extraction? I've looked for several articles that use this method, but they don't have the open source code.
Thanks again and best wishes.

ChongjianYUAN, 2022-11-15:
> Thank you very much for your excellent work. We are experimenting with your method, but we have encountered some problems. I noticed that you compared the depth continuous boundary with the depth [...]

Please refer to the Calibration::projection function and Calibration::edgeDetector function. You can get a depth image through projection function, then try the edgeDetector to get the depth-discontinuous boundary.

Xiaotianzhi, 2022-11-17:
> Please refer to the Calibration::projection function and Calibration::edgeDetector function. You can get a depth image through projection function, then try the edgeDetector to get the [...]

Thank you for your reply. I will take a look at these two functions and have a try . Before this, I have been reading Calibration: : LiDAREdgeExtraction but have no idea.

ChongjianYUAN, 2022-11-17:
> Thank you for your reply. I will take a look at these two functions and have a try . Before this, I have been reading Calibration: : LiDAREdgeExtraction but have no idea.

LiDAREdgeExtraction is used to extract the Depth-Continuous edge, not for depth-discontinuous edge.