hku-mars/livox_camera_calib #72: 作者您好！使用outdoor.yaml时发生的一个严重错误，以及一个无法在粗标定以后显示rviz的错误

lilrookie76, 2022-12-01:
您好，我在室外场景尝试使用标定，但是在使用config_outdoor.yaml进行标定时（也就是尝试过使用indoor yaml文件不会出现的错误），在迭代完成计算结果的时候会出现一个错误，如图1，导致无法输出结果：
[image]
另外不管使用哪种config的时候在粗标定结束后会无法正常使用rviz，报错如下：
[image]
希望得到您的回复，谢谢！

ChongjianYUAN, 2022-12-06:
> 您好，我在室外场景尝试使用标定，但是在使用config_outdoor.yaml进行标定时（也就是尝试过使用indoor yaml文件不会出现的错误），在迭代完成计算结果的时候会出现一个错误，如图1，导致无法输出结果： [image] 另外不管使用哪种config的时候在粗标定结束后会无法正常使用rviz，报错如下： [image] 希望得到您的回复，谢谢！

您好，rviz报错退出看起来像是内存不足的问题，至于第一个问题，暂时不知为何使用outdoor config会报错，可以调试尝试定位报错的代码具体在哪里

lilrookie76, 2022-12-10:
> > 您好，我在室外场景尝试使用标定，但是在使用config_outdoor.yaml进行标定时（也就是尝试过使用indoor yaml文件不会出现的错误），在迭代完成计算结果的时候会出现一个错误，如图1，导致无法输出结果： [image] 另外不管使用哪种config的时候在粗标定结束后会无法正常使用rviz，报错如下： [image] 希望得到您的回复，谢谢！
> 
> 您好，rviz报错退出看起来像是内存不足的问题，至于第一个问题，暂时不知为何使用outdoor config会报错，可以调试尝试定位报错的代码具体在哪里

好的好的，我这里使用您拍摄的户外标定照片就可以成功用outdoor config文件，所以应该是拍摄的标定场景问题，谢谢您的回答！