hku-mars/livox_camera_calib #79: catkin_make  ---> error

kt-rax, 2023-03-03:
root@ubuntu:~/catkin_ws# catkin_make
Base path: /root/catkin_ws
Source space: /root/catkin_ws/src
Build space: /root/catkin_ws/build
Devel space: /root/catkin_ws/devel
Install space: /root/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/root/catkin_ws/build"
####
####
#### Running command: "make -j32 -l32" in "/root/catkin_ws/build"
####
[  8%] Automatic MOC and UIC for target lidar_camera_multi_calib
[ 16%] Automatic MOC and UIC for target bag_to_pcd
[ 25%] Automatic MOC and UIC for target lidar_camera_calib
[ 25%] Built target lidar_camera_multi_calib_autogen
[ 25%] Built target bag_to_pcd_autogen
[ 25%] Built target lidar_camera_calib_autogen
[ 33%] Building CXX object livox_camera_calib/CMakeFiles/lidar_camera_multi_calib.dir/src/lidar_camera_multi_calib.cpp.o
[ 41%] Building CXX object livox_camera_calib/CMakeFiles/lidar_camera_calib.dir/src/lidar_camera_calib.cpp.o
[ 66%] Built target bag_to_pcd
In file included from /usr/include/eigen3/Eigen/Core:22:0,
                 from /root/catkin_ws/src/livox_camera_calib/include/common.h:3,
                 from /root/catkin_ws/src/livox_camera_calib/include/lidar_camera_calib.hpp:5,
                 from /root/catkin_ws/src/livox_camera_calib/src/lidar_camera_calib.cpp:1:
/usr/include/eigen3/Eigen/src/Core/util/ConfigureVectorization.h:38:38: warning: attribute ignored in declaration of ‘struct pcl::_PointXYZHSV’ [-Wattributes]
   #define EIGEN_ALIGN_TO_BOUNDARY(n) alignas(n)
                                      ^
/usr/include/eigen3/Eigen/src/Core/util/ConfigureVectorization.h:38:38: note: attribute for ‘struct pcl::_PointXYZHSV’ must follow the ‘struct’ keyword
In file included from /opt/ros/melodic/include/roslz4/lz4s.h:38:0,
                 from /opt/ros/melodic/include/rosbag/stream.h:46,
                 from /opt/ros/melodic/include/rosbag/chunked_file.h:46,
                 from /opt/ros/melodic/include/rosbag/bag.h:41,
                 from /root/catkin_ws/src/livox_camera_calib/include/lidar_camera_calib.hpp:25,
                 from /root/catkin_ws/src/livox_camera_calib/src/lidar_camera_calib.cpp:1:
/usr/include/lz4.h:196:57: error: conflicting declaration ‘typedef struct LZ4_stream_t LZ4_stream_t’
 typedef struct { long long table[LZ4_STREAMSIZE_U64]; } LZ4_stream_t;
                                                         ^~~~~~~~~~~~
In file included from /usr/include/flann/util/serialization.h:9:0,
                 from /usr/include/flann/util/matrix.h:35,
                 from /usr/include/flann/flann.hpp:41,
                 from /usr/include/pcl-1.8/pcl/kdtree/flann.h:50,
                 from /usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:45,
                 from /usr/include/pcl-1.8/pcl/search/kdtree.h:44,
                 from /usr/include/pcl-1.8/pcl/search/pcl_search.h:44,
                 from /usr/include/pcl-1.8/pcl/features/impl/feature.hpp:44,
                 from /usr/include/pcl-1.8/pcl/features/feature.h:498,
                 from /usr/include/pcl-1.8/pcl/features/normal_3d.h:44,
                 from /root/catkin_ws/src/livox_camera_calib/include/lidar_camera_calib.hpp:13,
                 from /root/catkin_ws/src/livox_camera_calib/src/lidar_camera_calib.cpp:1:
/usr/include/flann/ext/lz4.h:196:57: note: previous declaration as ‘typedef struct LZ4_stream_t LZ4_stream_t’
 typedef struct { long long table[LZ4_STREAMSIZE_U64]; } LZ4_stream_t;
                                                         ^~~~~~~~~~~~
In file included from /opt/ros/melodic/include/roslz4/lz4s.h:38:0,
                 from /opt/ros/melodic/include/rosbag/stream.h:46,
                 from /opt/ros/melodic/include/rosbag/chunked_file.h:46,
                 from /opt/ros/melodic/include/rosbag/bag.h:41,
                 from /root/catkin_ws/src/livox_camera_calib/include/lidar_camera_calib.hpp:25,
                 from /root/catkin_ws/src/livox_camera_calib/src/lidar_camera_calib.cpp:1:
/usr/include/lz4.h:249:72: error: conflicting declaration ‘typedef struct LZ4_streamDecode_t LZ4_streamDecode_t’
 typedef struct { unsigned long long table[LZ4_STREAMDECODESIZE_U64]; } LZ4_streamDecode_t;
                                                                        ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/flann/util/serialization.h:9:0,
                 from /usr/include/flann/util/matrix.h:35,
                 from /usr/include/flann/flann.hpp:41,
                 from /usr/include/pcl-1.8/pcl/kdtree/flann.h:50,
                 from /usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:45,
                 from /usr/include/pcl-1.8/pcl/search/kdtree.h:44,
                 from /usr/include/pcl-1.8/pcl/search/pcl_search.h:44,
                 from /usr/include/pcl-1.8/pcl/features/impl/feature.hpp:44,
[... 88 lines cut ...]
                 from /root/catkin_ws/src/livox_camera_calib/include/lidar_camera_calib.hpp:19,
                 from /root/catkin_ws/src/livox_camera_calib/src/lidar_camera_multi_calib.cpp:3:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a protected member of the SampleConsensusModel class [-Wdeprecated-declarations]
   SAC_SAMPLE_SIZE (sample_size_pairs, sample_size_pairs + sizeof (sample_size_pairs) / sizeof (SampleSizeModel));
   ^~~~~~~~~~~~~~~
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: note: declared here
livox_camera_calib/CMakeFiles/lidar_camera_calib.dir/build.make:89: recipe for target 'livox_camera_calib/CMakeFiles/lidar_camera_calib.dir/src/lidar_camera_calib.cpp.o' failed
make[2]: *** [livox_camera_calib/CMakeFiles/lidar_camera_calib.dir/src/lidar_camera_calib.cpp.o] Error 1
CMakeFiles/Makefile2:2500: recipe for target 'livox_camera_calib/CMakeFiles/lidar_camera_calib.dir/all' failed
make[1]: *** [livox_camera_calib/CMakeFiles/lidar_camera_calib.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
livox_camera_calib/CMakeFiles/lidar_camera_multi_calib.dir/build.make:89: recipe for target 'livox_camera_calib/CMakeFiles/lidar_camera_multi_calib.dir/src/lidar_camera_multi_calib.cpp.o' failed
make[2]: *** [livox_camera_calib/CMakeFiles/lidar_camera_multi_calib.dir/src/lidar_camera_multi_calib.cpp.o] Error 1
CMakeFiles/Makefile2:2526: recipe for target 'livox_camera_calib/CMakeFiles/lidar_camera_multi_calib.dir/all' failed
make[1]: *** [livox_camera_calib/CMakeFiles/lidar_camera_multi_calib.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j32 -l32" failed
root@ubuntu:~/catkin_ws# pwd

在catkin_ws与src目录下都放了项目源码livox_camera_calib 可以catkin_make但是有如上的错误
[image]
不然有错误：
The specified base path "/。。。 [contains] 。。。a package but "catkin_make" must be invoked in the root of workspace

kt-rax, 2023-03-03:
[image]

kt-rax, 2023-03-03:
ceres-solver用的哪个版本呢？谢谢

kt-rax, 2023-03-03:
[image]

Yangbit666, 2023-03-05:
> root@ubuntu:~/catkin_ws# catkin_make Base path: /root/catkin_ws Source space: /root/catkin_ws/src Build space: /root/catkin_ws/build Devel space: /root/catkin_ws/devel Install space: [...]

Yangbit666, 2023-03-05:
你好，遇到了与上边相同的问题，想请问你解决了吗？
/usr/include/lz4.h:249:72: error: conflicting declaration ‘typedef struct LZ4_streamDecode_t LZ4_streamDecode_t’
 typedef struct { unsigned long long table[LZ4_STREAMDECODESIZE_U64]; } LZ4_streamDecode_t;
                                                                        ^~~~~~~~~~~~~~~~~~
In file included from /usr/include/flann/util/serialization.h:9:0,
                 from /usr/include/flann/util/matrix.h:35,
                 from /usr/include/flann/flann.hpp:41,
                 from /usr/include/pcl-1.8/pcl/kdtree/flann.h:50,
                 from /usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:45,
                 from /usr/include/pcl-1.8/pcl/search/kdtree.h:44,
                 from /usr/include/pcl-1.8/pcl/search/pcl_search.h:44,
                 from /usr/include/pcl-1.8/pcl/features/impl/feature.hpp:44,
                 from /usr/include/pcl-1.8/pcl/features/feature.h:498,
                 from /usr/include/pcl-1.8/pcl/features/normal_3d.h:44,
                 from /home/yanglin/catkin_ws/src/livox_camera_calib/include/lidar_camera_calib.hpp:13,
                 from /home/yanglin/catkin_ws/src/livox_camera_calib/src/lidar_camera_calib.cpp:1:
/usr/include/flann/ext/lz4.h:249:72: note: previous declaration as ‘typedef struct LZ4_streamDecode_t LZ4_streamDecode_t’
 typedef struct { unsigned long long table[LZ4_STREAMDECODESIZE_U64]; } LZ4_streamDecode_t;
                                                                        ^~~~~~~~~~~~~~~~~~
In file included from /opt/ros/melodic/include/roslz4/lz4s.h:38:0,
                 from /opt/ros/melodic/include/rosbag/stream.h:46,
                 from /opt/ros/melodic/include/rosbag/chunked_file.h:46,
                 from /opt/ros/melodic/include/rosbag/bag.h:41,
                 from /home/yanglin/catkin_ws/src/livox_camera_calib/include/lidar_camera_calib.hpp:25,
                 from /home/yanglin/catkin_ws/src/livox_camera_calib/src/lidar_camera_multi_calib.cpp:3:
/usr/include/lz4.h:196:57: error: conflicting declaration ‘typedef struct LZ4_stream_t LZ4_stream_t’
 typedef struct { long long table[LZ4_STREAMSIZE_U64]; } LZ4_stream_t;
                                                         ^~~~~~~~~~~~
In file included from /usr/include/flann/util/serialization.h:9:0,
                 from /usr/include/flann/util/matrix.h:35,
                 from /usr/include/flann/flann.hpp:41,
                 from /usr/include/pcl-1.8/pcl/kdtree/flann.h:50,
                 from /usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:45,
                 from /usr/include/pcl-1.8/pcl/search/kdtree.h:44,
                 from /usr/include/pcl-1.8/pcl/search/pcl_search.h:44,
                 from /usr/include/pcl-1.8/pcl/features/impl/feature.hpp:44,
                 from /usr/include/pcl-1.8/pcl/features/feature.h:498,
                 from /usr/include/pcl-1.8/pcl/features/normal_3d.h:44,
                 from /home/yanglin/catkin_ws/src/livox_camera_calib/include/lidar_camera_calib.hpp:13,
                 from /home/yanglin/catkin_ws/src/livox_camera_calib/src/lidar_camera_multi_calib.cpp:3:
/usr/include/flann/ext/lz4.h:196:57: note: previous declaration as ‘typedef struct LZ4_stream_t LZ4_stream_t’
 typedef struct { long long table[LZ4_STREAMSIZE_U64]; } LZ4_stream_t;
                                                         ^~~~~~~~~~~~
In file included from /opt/ros/melodic/include/roslz4/lz4s.h:38:0,
                 from /opt/ros/melodic/include/rosbag/stream.h:46,
                 from /opt/ros/melodic/include/rosbag/chunked_file.h:46,
                 from /opt/ros/melodic/include/rosbag/bag.h:41,
                 from /home/yanglin/catkin_ws/src/livox_camera_calib/include/lidar_camera_calib.hpp:25,
                 from /home/yanglin/catkin_ws/src/livox_camera_calib/src/lidar_camera_multi_calib.cpp:3:
/usr/include/lz4.h:249:72: error: conflicting declaration ‘typedef struct LZ4_streamDecode_t LZ4_streamDecode_t’
 typedef struct { unsigned long long table[LZ4_STREAMDECODESIZE_U64]; } LZ4_streamDecode_t;
                                                                        ^~~~~~~~~~~~~~~~~~