hku-mars/livox_camera_calib #80: The visual result is abnormal？

kt-rax, 2023-03-03:
I have follow the steps and up to the roslaunch.

But seems got no result.Is there anybody could give some suggestions or comments.Thanks!

roslaunch livox_camera_calib calib.launch

[image]

kt-rax, 2023-03-03:
[image]

Yangbit666, 2023-03-05:
前辈你好，想请问您编译报错的问题解决了吗？

Yangbit666, 2023-03-05:
> 

Maybe I need to use lower version Ceres Soler,  as other poster metioned.

Snowbound9173, 2023-08-25:
Looks like you didn't correctly configure the paths required. You can even see the message "Failed to open settings for file at: ...."

You will need to go into  the calib.yaml file and update your paths. I realise this is super old but if someone stumbles across it it might help.

```
# Data path. adjust them!
common:
    image_file: "/home/livox/ws_livox_camera_calib/data/image/0.png"
    pcd_file: "/home/livox/ws_livox_camera_calib/data/pcd/0.pcd"
    result_file: "/home/livox/ws_livox_camera_calib/data/output/extrinsic.txt"

# Camera Parameters. Adjust them!
```