hku-mars/livox_camera_calib #84: 关于使用livox mid360与相机进行标定

LebinLiang, 2023-03-26:
我这边尝试使用mid360和海康威视的相机使用该功能包进行外参标定，但是mid360的点云fov范围很大，相机只是一个普通的广角约70度fov。我按着readme的方法进行测试，最终无法得到结果，想问一下这种情况应该如何处理呢？

ChongjianYUAN, 2023-03-28:
> 我这边尝试使用mid360和海康威视的相机使用该功能包进行外参标定，但是mid360的点云fov范围很大，相机只是一个普通的广角约70度fov。我按着readme的方法进行测试，最终无法得到结果，想问一下这种情况应该如何处理呢？

雷达FOV比相机大的话，应该不会标定失败，无法得到结果是指得到的外参差还是无法进行到优化步骤呢？

kong9999, 2023-04-10:
你好，我使用mid360和realsense435做标定，但是mid360的点云很稀疏，和相机标定无法得到结果。一直在报错Cannot create a KDTree with an empty input cloud! 我根据之前的方法都做了修改，但是还是不行，是不是因为mid360的点云太稀疏，没有办法提取到边缘的问题。

[image]

ChongjianYUAN, 2023-04-20:
> 

Mid360点云很稀疏的话，可以跑一下FAST-LIO算法先积累稠密的点云，再和初始位姿下的图像一起进行标定

GavinGaoYan, 2024-02-23:
订阅 后续有无详细介绍

WeiZixu-HIT, 2024-08-08:
> @ChongjianYUAN 您好，我使用mid360和realsense435做标定，但是mid360的点云很稀疏，和相机标定无法得到外参且也进入不到优化步骤。一直在报错Cannot create a KDTree with an empty input cloud! 我根据之前的方法都做了修改，但是还是不行，是不是因为mid360的点云太稀疏，没有办法提取到边缘的问题。
> 
> [image]
> 
> [image] [image]

请问您用mid360和d435标定成功了吗