hku-mars/livox_camera_calib #85: 运行软件出错，加载数据问题

completezealous, 2023-04-16:
- 我用的是你们云盘上的数据，运行软件时，黑屏，报错，
-这是我设置的文件路径 common:
    image_file: "/home/lk/catkin_ws/src/livox_camera_calib/data/single_scene_calibration/0.png"
    pcd_file: "/home/lk/catkin_ws/src/livox_camera_calib/data/single_scene_calibration/0.pcd"
    result_file: "/home/lk/catkin_ws/src/livox_camera_calib/result/extrinsic.txt"

# Camera Parameters. Adjust them!
camera:
    camera_matrix: [1364.45, 0.0,      958.327,
                0.0,     1366.46,  535.074,
                0.0,     0.0,      1.0     ]
    dist_coeffs: [0.0958277, -0.198233, -0.000147133, -0.000430056, 0.000000]

# Calibration Parameters.!
calib:
    calib_config_file: "/home/lk/catkin_ws/src/livox_camera_calib/config/config_outdoor.yaml"
    use_rough_calib: true # set true if your initial_extrinsic is bad

“(base) lk@sun:~/catkin_ws$ roslaunch livox_camera_calib calib.launch
... logging to /home/lk/.ros/log/8bfc51c8-dc14-11ed-9ebe-0cc47a6c9e2b/roslaunch-sun-26079.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sun:32971/

SUMMARY
========

PARAMETERS
 * /calib/calib_config_file: /home/lk/catkin_w...
 * /calib/use_rough_calib: True
 * /camera/camera_matrix: [1364.45, 0.0, 95...
 * /camera/dist_coeffs: [0.0958277, -0.19...
 * /common/image_file: /home/lk/catkin_w...
 * /common/pcd_file: /home/lk/catkin_w...
 * /common/result_file: /home/lk/catkin_w...
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    lidar_camera_calib (livox_camera_calib/lidar_camera_calib)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [26093]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8bfc51c8-dc14-11ed-9ebe-0cc47a6c9e2b
process[rosout-1]: started with pid [26104]
started core service [/rosout]
process[lidar_camera_calib-2]: started with pid [26108]
process[rviz-3]: started with pid [26112]
pcd_file path:/home/lk/catkin_ws/src/livox_camera_calib/data/single_scene_calibration/0.pcd
[lidar_camera_calib-2] process has died [pid 26108, exit code -11, cmd /home/lk/catkin_ws/devel/lib/livox_camera_calib/lidar_camera_calib __name:=lidar_camera_calib __log:=/home/lk/.ros/log/8bfc51c8-dc14-11ed-9ebe-0cc47a6c9e2b/lidar_camera_calib-2.log].
log file: /home/lk/.ros/log/8bfc51c8-dc14-11ed-9ebe-0cc47a6c9e2b/lidar_camera_calib-2*.log”

其中在软件界面显示未接收到图片

completezealous, 2023-04-16:
[image]

completezealous, 2023-04-16:
[image]

zjutcaicai, 2023-05-10:
请问你解决了吗，我也这个问题

completezealous, 2023-05-10:
没有解决  我换了一个服务器账户   没有出现这种问题

Goughhhhhh, 2023-07-28:
> 请问你解决了吗，我也这个问题
我也遇到了这个问题，请问您解决了吗？

completezealous, 2023-07-28:
> > 请问你解决了吗，我也这个问题
> > 我也遇到了这个问题，请问您解决了吗？

没有解决   换了一个服务器账户 就没出现这种问题 可能是opencv版本问题

Goughhhhhh, 2023-07-30:
> > > 请问你解决了吗，我也这个问题
> > > 我也遇到了这个问题，请问您解决了吗？
> 
> 没有解决 换了一个服务器账户 就没出现这种问题 可能是opencv版本问题

那您在成功运行的服务器中用的是什么版本的OpenCV呢？

completezealous, 2023-07-31:
> > > > 请问你解决了吗，我也这个问题
> > > > 我也遇到了这个问题，请问您解决了吗？
> > 
> > 
> > 没有解决 换了一个服务器账户 就没出现这种问题 可能是opencv版本问题
> 
> 那您在成功运行的服务器中用的是什么版本的OpenCV呢？

3.4.16   当然也有可能是你点云文件问题 点云文件距离太长  也会报错  一般需要裁剪一下

Wayne-Ning, 2024-12-04:
> > 请问你解决了吗，我也这个问题
> > 我也遇到了这个问题，请问您解决了吗？

请问你解决这个问题了吗？
我遇到的也是这个问题