hku-mars/livox_camera_calib #86: 咨询标定精度不足

MengHuiLei, 2023-04-23:
非常感谢您发布这么好的代码，效果很棒！！！！我最近在测试时遇到一个问题，向您咨询一下。
硬件：海康相机+horizon，选取室内场景进行标定。（选取多个室内场景，运行multi_calib.launch）
运行完，输出图片：（看着匹配度还是很高的）
[image]
）
问题：平移矩阵中x很准确，y、z精度不足，大概差10+cm。（算法输出：0.160863, -0.073921, -0.180723）
[image]
您知道问题大概出在哪吗？（相机标定矩阵，采用实物距离进行测试，精度在2cm）

MengHuiLei, 2023-04-23:
@ChongjianYUAN

ChongjianYUAN, 2023-04-23:
> 非常感谢您发布这么好的代码，效果很棒！！！！我最近在测试时遇到一个问题，向您咨询一下。 硬件：海康相机+horizon，选取室内场景进行标定。（选取多个室内场景，运行multi_calib.launch） 运行完，输出图片：（看着匹配度还是很高的） [image] ） 问题：平移矩阵中x很准确，y、z精度不足，大概差10+cm。（算法输出：0.160863, -0.073921, -0.180723） [image] 您知道问题大概出在哪吗？（相机标定矩阵，采用实物距离进行测试，精度在2cm）

感谢您的测试；这是我的一些理解: 平移矩阵中x比y准的话，大概率是场景中的纵向边缘(约束x)比横向边缘(约束y)数量多，可以尝试把设备倒置90度在场景中采集些数据参与标定。至于z的话，由论文里面的公式20可知，z轴的雅克比是最小的，所以标定出来的结果里z的方差是最大的，这点比较难以避免，只能尝试使用更多的场景一起标定；此外内参的精确度也一定影响到外参，使用cad模型给的外参有时候投影效果也并不好

MengHuiLei, 2023-04-24:
搞定，z方向精度也达到2cm以内。非常感谢。。。。

closed 2023-04-24 by MengHuiLei

MengHuiLei, 2023-05-10:
> @MengHuiLei 您好，想请教一下您的x，y，z精度是怎么求的呢

其实我没有严格的真值，把硬件构建时的三维图纸作为参考真值。

tdcsu, 2024-05-16:
> 搞定，z方向精度也达到2cm以内。非常感谢。。。。

感谢，有帮助！