hku-mars/livox_camera_calib #87: 硬件同步咨询

MengHuiLei, 2023-04-27:
@ChongjianYUAN 
想复现fast_livo的效果，测试时遇到一点问题，想向您咨询一下。
前提：
1. 采用[Handheld_ws](https://github.com/xuankuzcr/Handheld_ws)这里的时间同步器，稳定发布1Hz和10HzPPS信号源。
2. 相机：mv-cs050-10gc-pro，雷达：horizon，采用千兆交换机与笔记本相连；
3. 确定相机和雷达均能收到PPS信号；
不同点：
1. 由于相机的接口方式不同，故未能采用Handheld_ws中的相机驱动，但同样采用 接收到相机数据的时间 赋值 相机话题的时间；

出现的问题：采用Handheld_ws中的livox_ros_driver和时间同步器，发现相机和雷达的时间对应不上 且 雷达的时间较为滞后。
如图：图中左侧为传感器发布的时间，右侧为系统时间
[image]

MengHuiLei, 2023-04-27:
另外该livox_ros_driver在pps模式下，fast-lio建图失败。

MengHuiLei, 2023-04-28:
@ChongjianYUAN

ChongjianYUAN, 2023-05-06:
> @ChongjianYUAN 想复现fast_livo的效果，测试时遇到一点问题，想向您咨询一下。 前提：
> 
> 1. 采用[Handheld_ws](https://github.com/xuankuzcr/Handheld_ws)这里的时间同步器，稳定发布1Hz和10HzPPS信号源。
> 2. 相机：mv-cs050-10gc-pro，雷达：horizon，采用千兆交换机与笔记本相连；
> 3. 确定相机和雷达均能收到PPS信号；
>    不同点：
> 4. 由于相机的接口方式不同，故未能采用Handheld_ws中的相机驱动，但同样采用 接收到相机数据的时间 赋值 相机话题的时间；
> 
> 出现的问题：采用Handheld_ws中的livox_ros_driver和时间同步器，发现相机和雷达的时间对应不上 且 雷达的时间较为滞后。 如图：图中左侧为传感器发布的时间，右侧为系统时间 [image]

Fast-livo应该不支持不同步的雷达和相机，只有硬同步才能保证雷达和相机时间一致；至于livox_ros_driver在pps模式下，fast-lio建图失败的问题，应该是pps模式下IMU时间戳和LiDAR时间戳没对齐，可以参考livox_ros_drive驱动的6. livox_ros_driver timestamp synchronization function

MengHuiLei, 2023-05-10:
已解决

lalawinter, 2024-08-12:
你好，我也想用horizon+海康的 相机复现，能否加个联系方式交流下硬件问题，我的qq是1538249766

lalawinter, 2024-08-12:
还有一个问题就是用horizon复现效果怎么样相比于avia