hku-mars/livox_camera_calib #89: install on ubuntu 20.04, can I use ROS Noetic ?

AlmogHagag, 2023-05-14:
install on ubuntu 20.04, can I use ROS Noetic ?
Thanks

AlmogHagag, 2023-05-14:
[image]

Can anyone help with that?
Thanks,

AlmogHagag, 2023-05-16:
clone ceres and compile branch  1.14.x, work for me. Good luck

Chandan-kumar-R, 2024-07-12:
> ![image](https://private-user-images.githubusercontent.com/6381324/238193003-78a12e2d-c9ad-4b10-bede-e9204b801fdb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]
replace with these and it will works:
      ceres::Manifold *q_parameterization =
          new ceres::EigenQuaternionManifold();