hku-mars/livox_camera_calib #92: Demo运行时，launch文件存在opencv报错

LN-123-hub, 2023-07-08:
[image]

LN-123-hub, 2023-07-08:
请问大家有遇到这问题吗

Yajiang, 2023-07-19:
> 请问大家有遇到这问题吗

或许你已经解决了，我遇到了相同的问题，因为我的pcd文件中不包含intensity值，lidar_camera_calib.hpp 中这段代码改掉就好
[image]

czy1998916, 2024-07-22:
@Yajiang 你好，我也出现了类似的错误，一直无处下手，请问你能帮忙看看吗
[image]

zwzzxxh, 2025-05-03:
> [@Yajiang](https://github.com/Yajiang) 你好，我也出现了类似的错误，一直无处下手，请问你能帮忙看看吗 [image]

你好，我遇到了同样的问题，请问你们解决了吗