hku-mars/livox_camera_calib #99: catkin_make error

ezthor, 2023-11-18:
i can run the demo successfully once,but maybe after i use pip -r to install an requirement list in another anaconda environment and change the version of pcl ,i encounter this problem:
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/9/../../../x86_64-linux-gnu/libpcl_io.so: undefined reference to `libusb_set_option'
.Now I am using Ubuntu 20.04 and pcl version : libpcl-dev     1.10.0+dfsg-5ubuntu1
is it a problem with pcl version? which version should i use? plz help,thanks!!
[image]

ezthor, 2023-12-23:
I used docker to reconfigure the environment and successfully catkin_make ,though still don't know how to fix this problem.

WilsonGuo, 2024-05-09:
问题出现在安装了海康摄像头的配置软件，删除这个软件即可正常编译和运行程序。

分析问题：这个错误是在我安装完海康摄像头和速腾的SDK后，就出现的，所以问题大概率出现这里，然后直接全局搜索 libusb 的库文件，然后在海康的MVS下找到这个文件了，然后把MVS删除后，就可以编译了。

TZECHIN6, 2024-08-12:
但刪掉後還能用到HIK相機嗎? 還是只需刪`libusb-x.so.x`

ezthor, 2024-08-16:
> 问题出现在安装了海康摄像头的配置软件，删除这个软件即可正常编译和运行程序。
> 
> 分析问题：这个错误是在我安装完海康摄像头和速腾的SDK后，就出现的，所以问题大概率出现这里，然后直接全局搜索 libusb 的库文件，然后在海康的MVS下找到这个文件了，然后把MVS删除后，就可以编译了。

可能是这个原因，我也用了海康驱动，后面在docker里配置环境不需要海康驱动所以没问题

ezthor, 2024-08-16:
> 但刪掉後還能用到HIK相機嗎? 還是只需刪`libusb-x.so.x`

这我

ezthor, 2024-08-16:
> 但刪掉後還能用到HIK相機嗎? 還是只需刪`libusb-x.so.x`

这我不太清楚，但我建议外部环境保留MVS在docker中重新配置一个标定环境。

closed 2024-08-16 by ezthor