hku-mars/livox_camera_calib #100: 运行自己的数据时，出现Matrix operand is an empty matrix. in function 'checkOperandsExist'报错

ezthor, 2023-12-23:
运行自己的数据时，出现报错：
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) ../modules/core/src/matrix_expressions.cpp:23: error: (-5:Bad argument) Matrix operand is an empty matrix. in function 'checkOperandsExist'
我是用livox viewer将mid70录制的csv格式通过自己的脚本仿照官方demo中的pcd进行转换的，然后报错了。我一开始发现存在x，y，z为0时，intensity不为0的数据，而我看官方的是intensity均为0，所以我将其intensity转为0了，然后还是报错，我就将x，y，z为0的点直接删除，重新统计点的个数，现在还是报错，我想知道这部分点转换时该如何处理？报错是因为这部分的原因吗？
需要补充的是，我保存的csv格式好像是积分了几秒之后的点云图，所以保存了时间戳信息，我现在直接将总的点云去掉时间信息直接转为pcd格式会有问题吗？
我的csv文件格式如下：
[image]
我的转换后pcd文件如下，发现好像有重复点，会有问题吗？：
[image]
我的yaml配置如下：
[image]
麻烦各位大佬救救，谢谢

ezthor, 2023-12-23:
发错了，那个pcd是官方的，我的pcd是这个：
[image]

closed 2023-12-23 by ezthor

HelloSZS, 2024-05-31:
请问这个应该怎么解决嘛？