hku-mars/livox_camera_calib #104: Cali with MID360 problem..

eung-ark, 2024-03-18:
Hello,
I am enjoying your code. 
However, when I try to calibrate with MID360, the following phenomenon occurs. 

[image]

I think I configured my build correctly, but the PCD data is not what I expected.
 I need help.

Lavinci, 2024-03-30:
Hello, I encountered the same problem as you. I finally found out that my camera internal parameters were wrong. You should focus on checking your dist_coeffs. good luck!

WeiZixu-HIT, 2024-08-06:
> Hello, I encountered the same problem as you. I finally found out that my camera internal parameters were wrong. You should focus on checking your dist_coeffs. good luck!

你好，我标定相机内参好多次，都是差不多的结果，但是每次用来标定MID360的效果就很差，请问是相机内参的原因吗？

AAAzelda, 2024-08-19:
> > 你好，我遇到了和你一样的问题。我终于发现我的相机内部参数是错误的。你应该专注于检查你的dist_coeffs。祝你好运！
> 
> 你好，我标定相机内参好多次，都是差不多的结果，但是每次用来标定MID360的效果就很差，请问是相机内参的原因吗？

你好，我和你问题一样，请问你有找到原因吗？

SanghyunPark01, 2025-11-03:
https://github.com/SanghyunPark01/livox_camera_calib

use this repo. It contains pointcloud cropping for MID360