hku-mars/livox_camera_calib #112: 线特征误匹配

Geralt-of-winterfall, 2024-07-23:
代码中似乎没有针对场景中线特征较为丰富的场景出现的较多线特征误匹配较多的情况作额外处理。
是否可以针对投影线特征作更多的筛选和滤除？例如柱子背后的线特征也被投影，可能会造成误匹配。
线特征匹配是否可以利用更多属性进行匹配？

czy1998916, 2024-07-24:
[image]
n你有遇到这种情况吗？我用的是速腾16线激光雷达和usb相机

Geralt-of-winterfall, 2024-07-24:
我遇到的是cv::projectPoints函数的问题，投影到图像平面的pts_3d为空，cv报错。
如果用的是单帧点云的话，16线点云密度太低了，提取不到线特征。

czy1998916, 2024-07-24:
@Geralt-of-winterfall 是的，所以我用了fastlio来进行特征提取，然后就出现上图报错

Geralt-of-winterfall, 2024-07-25:
[cv::projectPoints](https://github.com/hku-mars/livox_camera_calib/blame/061fdaa647fc806e59d73a8505a05a10dfcfdaa1/include/lidar_camera_calib.hpp#L320)，大概率是这个函数的问题，检查一下输入的参数3dpts是否为空

czy1998916, 2024-07-30:
@Geralt-of-winterfall 这个问题结局后点云对不上图像，请问你正常运行了吗，能加个联系方式沟通一下吗，我qq1781113105，十分感谢
[image]