hku-mars/livox_camera_calib #113: 如果要使用自己的数据，应该怎么解决一些报错，比如相机内参部分的错误

happypuppy128, 2024-08-07:
[image]
因为暂时是测试阶段，当时采集相机图片就是使用的手机拍摄的，手机的相机内参矩阵我没办法得知，使用棋盘格标定出来的还是会报错，希望能有大佬看到，帮忙解答一下，或者有没有办法避免这个错误

zl1344, 2024-08-08:
我也遇到了这个问题，原因是你的PCD点云没有intensity这个属性

happypuppy128, 2024-08-08:
> 我也遇到了这个问题，原因是你的PCD点云没有intensity这个属性

请问能加个联系方式一起讨论一下吗，万分感谢QQ: 1524056416

happypuppy128, 2024-08-08:
> 我也遇到了这个问题，原因是你的PCD点云没有intensity这个属性

我使用的是livox mid360这个型号的，我应该如何操作，确保我也有这个属性呢，请问您有解决嘛？

zl1344, 2024-08-08:
我也是用的MID360，参考这个博客的获取PCD文件的格式，可以跑通，但是效果不是很好，https://blog.csdn.net/a6666686678/article/details/138096601

happypuppy128, 2024-08-08:
> 我也是用的MID360，参考这个博客的获取PCD文件的格式，可以跑通，但是效果不是很好，https://blog.csdn.net/a6666686678/article/details/138096601

十分感谢你的帮助，我晚点好好看这个博客，希望能跑通

happypuppy128, 2024-08-08:
> 我也遇到了这个问题，原因是你的PCD点云没有intensity这个属性

冒昧再问一下，大佬请问你说的这个属性是需要去改livox雷达驱动的配置文件解决，还是说更改作者的这个项目的某些部分吗?，请您明示一下，万分感谢。

zl1344, 2024-08-09:
什么也不用改，就正常录制一个bag包，用那个博客的方法生成PCD文件就行

happypuppy128, 2024-08-09:
> 什么也不用改，就正常录制一个bag包，用那个博客的方法生成PCD文件就行

我之前采集的数据能用吗，使用的lvx2后缀保存的点云数据，因为数据特殊短时间没办法再次采集，自带的viewer2这个软件似乎不支持直接转bag包，有什么不错的办法吗，谢谢您的耐心回复

happypuppy128, 2024-08-09:
> 什么也不用改，就正常录制一个bag包，用那个博客的方法生成PCD文件就行

[image]

happypuppy128, 2024-08-09:
> 什么也不用改，就正常录制一个bag包，用那个博客的方法生成PCD文件就行

因为数据是之前使用livox viewer2采集的，似乎不涉及修改livox_ros_driver下的config/livox_lidar_config.json,我的意思是，那么是不是不需要再重新采集，只需要将lvx2的文件转成rosbag文件就行了？

[image]

zl1344, 2024-08-09:
我直接录的bag包，没用过lvx2

happypuppy128, 2024-08-09:
> 我直接录的bag包，没用过lvx2

十分感谢你的帮助，我成功运行了自己的数据，如你所言，效果并不是很好，大佬你当时是怎么解决的，是有别的开源思路吗？

happypuppy128, 2024-08-09:
> 我直接录的bag包，没用过lvx2

十分感谢你的帮助，我成功运行了自己的数据，如你所言，效果并不是很好，大佬你当时是怎么解决的，有什么别的开源的方法吗？

zhangyuran-gg, 2024-10-17:
> intensity

你好，请问你解决这个问题了吗