hku-mars/livox_camera_calib #119: catkin_make error

phdh4, 2024-09-24:
When I started build and run catkin_make in terminal, I met problem as following:
[image]
[image]
I tried different version of Ceres but this error still couldn't be solved. 
My System is Ubuntu18.04 and Ceres version is 2.0.0.

zhangyuran-gg, 2024-10-12:
> 当我在终端中开始构建和运行 catkin_make 时，我遇到了如下问题： ![01](https://private-user-images.githubusercontent.com/99808815/370142462-6c75b583-8aef-4872-86cf-764be8ccc089.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3MjExMjAsIm5iZiI6MTcyODcyMDgyMCwicGF0aCI6Ii85OTgwODgxNS8zNzAxNDI0NjItNmM3NWI1ODMtOGFlZi00ODcyLTg2Y2YtNzY0YmU4Y2NjMDg5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEyVDA4MTM0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZiOGZlNzQ4YWZiMWNiMGQ0NzUyNWVmMTNkNWYwZjI4MTM4ZWM2OWYxNGJlZDBkYzhhODUzMDhjMDE5NTZmMGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.FVPxELl5i2bgNO9A03JsRierX8AfgxQjbOjxS8xNFU4) ![02](https://private-user-images.githubusercontent.com/99808815/370143739-73b0924f-336e-4b25-bb4d-46c05f72e412.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg3MjExMjAsIm5iZiI6MTcyODcyMDgyMCwicGF0aCI6Ii85OTgwODgxNS8zNzAxNDM3MzktNzNiMDkyNGYtMzM2ZS00YjI1LWJiNGQtNDZjMDVmNzJlNDEyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDEyVDA4MTM0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE4MjNlNzEwMjQzOGE0OWEyMzkyYzA4ZmI2YzIyOTlkMmYwZWVhMmM5MDRhOGRhYTQ5NzkxODMyYjA5MjE0YTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.n6zGkF9R3oxHGwhvwMqroXJop8arfk4akxJ3J29SHtU)我尝试了不同版本的 Ceres，但这个错误仍然无法解决。我的系统是 Ubuntu18.04，Ceres 版本是 2.0.0。

我用的ceres2.2.0，编译时同样出现ceres版本问题，请问你解决了吗：
/home/zyr/lccalib_ws/src/livox_camera_calib/src/lidar_camera_multi_calib.cpp:313:15: error: expected type-specifier
  313 |           new ceres::EigenQuaternionParameterization();

phdh4, 2024-10-23:
> 我用的ceres2.2.0，编译时同样出现ceres版本问题，请问你解决了吗： /home/zyr/lccalib_ws/src/livox_camera_calib/src/lidar_camera_multi_calib.cpp:313:15: error: expected type-specifier 313 | new [...]

ceres2.2.0不行，安装更低的版本，2.0.0可以，安装更低版本时请完全卸载ceres2.2.0。

XuJ1E, 2024-11-12:
> When I started build and run catkin_make in terminal, I met problem as following: ![01](https://private-user-images.githubusercontent.com/99808815/370142462-6c75b583-8aef-4872-86cf-764be8ccc089.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzNzUxNzksIm5iZiI6MTczMTM3NDg3OSwicGF0aCI6Ii85OTgwODgxNS8zNzAxNDI0NjItNmM3NWI1ODMtOGFlZi00ODcyLTg2Y2YtNzY0YmU4Y2NjMDg5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEyVDAxMjc1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM2NWUyNjMzNDk0NDA3NWJjY2RlNjk0M2UxODM1OGNjYzg3MDYxZTYxYzMyYzkzNTA2ZGFiYmI4MWIyZDk5MmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ga9iqcuc7PhPkkBmNEz9a8IsR3Ce8ocjpjj_2GOon2k) ![02](https://private-user-images.githubusercontent.com/99808815/370143739-73b0924f-336e-4b25-bb4d-46c05f72e412.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEzNzUxNzksIm5iZiI6MTczMTM3NDg3OSwicGF0aCI6Ii85OTgwODgxNS8zNzAxNDM3MzktNzNiMDkyNGYtMzM2ZS00YjI1LWJiNGQtNDZjMDVmNzJlNDEyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTEyVDAxMjc1OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRkMWViOGRkODBmNDVlNjZhZjhlN2VkMjQ5NjE1M2EzYTE3Zjc0YmMwMmY5ODI0MmJhYTI2NDRlOThlNmQzYTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.WslDjVMh_XxRxjmltgQIURpG22k6-GJbo4MkSLCkwG8) I tried different version of Ceres but this error still couldn't be solved. My System is Ubuntu18.04 and Ceres version is 2.0.0.

======================================
1、download from url :http://ceres-solver.org/ceres-solver-2.0.0.tar.gz
2、mkdir build
       cd build
      cmake -D CMAKE_INSTALL_PREFIX=/usr/local/ceres .. #The premise is to create a 'ceres' file yourself under '/usr/local'
      sudo make install

85256638, 2024-11-14:
不用卸载CERES 2.2.0重装2.0.0直接改code：

/catkin_ws/src/livox_camera_calib/src 下的两个文件： 

[image]

把之前的 ceres::LocalParameterization *q_parameterization = new ceres::EigenQuaternionParameterization();

改成  ceres::LocalParameterization *q_parameterization = new ceres::EigenQuaternionManifold();

再catkin_make

XuJ1E, 2024-11-14:
> @zhangyuran-gg @phdh4 
> 不用卸载CERES 2.2.0重装2.0.0直接改code：
> 
> /catkin_ws/src/livox_camera_calib/src 下的两个文件： 
> 
> [image]
> 
> 
> 把之前的 `ceres::LocalParameterization *q_parameterization = new ceres::EigenQuaternionParameterization();`
> 
> 改成  `ceres::LocalParameterization *q_parameterization = new ceres::EigenQuaternionManifold();`
> 
> 再catkin_make
> 
> 
> 参考以下：
> [https://github.com/colmap/colmap/issues/1475](url)
> [https://blog.csdn.net/zb0824zb/article/details/139421686](url)
>